vlachoudis/bCNC #1542: Separate Plugin into data entry and plugin itself

MARIOBASZ, 2021-02-14:
I am reasoning well. Is what I write silly?
A plugin may need to summon another plugin. In the current way you have to rewrite code.
Example: If you wanted to use a trochoidal strategy for cajeado, or cajeado with islands, do not rewrite the double path on which the trochoids develop.
If I need to use spiral for ur enlarging a hole, too.
The drill plug-in more neatly can be used for drilling, tapping, countersinking.

Harvie, 2021-02-14:
Yes. This can be done... It will be probably needed soon.

DodoLaSaumure, 2021-02-14:
@MARIOBASZ 
Not sure to understand
Do you want to call a class from another plugin ?

MARIOBASZ, 2021-02-14:
Yes. Call plugin from other plugin

DodoLaSaumure, 2021-02-14:
@MARIOBASZ 
I am not an architect, but a friend has confirmed to me that it is not a good idea to call a plugin from another.
Plugins in python is a usual architecture, where plugins are supposed to stay independent each other.
If you need to call common code, please put the parent class inside a lib, and call this lib from derived plugins.
This how plugins are supposed to work.
if plugins call plugins, we will soon get spaghetti code.
Pierre

DodoLaSaumure, 2021-02-15:
You are welcome @MARIOBASZ 
To better understand the concept of plugin, a plugin is like a device you would add to a main device to add extra feature.
The main device has to be able to work with or without the plugin you add to it.
And if you plug other devices, they are not supposed to interfere they have to work independently from each other.

I would much prefer duplicated code rather than calling plugins from main app or calling plugins from plugin.

Pierre

MARIOBASZ, 2021-02-15:
Understood, Pierre.
Anyway, I see both advantages in having the plugin in a single file inside the plugins folder: simplification, keeping usual logic.
I also continue to see advantages in having, within the plugins folder, a folder with two files: plugin_xx_parameters (data entry) and plugin_xx_plugin (executes).
I have done the usual way for trochoid 3d, writing trochoid_path inside cnc.py to be summoned.
In some cases, I understand that not the most usual, to some new complement
it may be useful to make use of a plugin_xx_plugin functionality.
Or more than one.
I understand the concept of what you tell me, maybe it is the correct or more practical way.
Hugs, Mario.