vlachoudis/bCNC #1550: How to configure a dummy milling machine?

treki1, 2021-02-20:
So that I can test the G-Code on another PC, do I want to configure a dummy milling machine?
How can I do that?

Harvie, 2021-02-20:
Use g-code simulator such as https://camotics.org/

treki1, 2021-02-20:
Thank you.
But I want to check the G-Code for compatibility with bCNC.

Harvie, 2021-02-21:
>  But I want to check the G-Code for compatibility with bCNC.

bCNC does not interpret g-code. bCNC only shows preview of the g-code on the screen. But it is ultimately interpreted by GRBL. so when GRBL does not support your g-code it will fail no matter what bCNC does.

paulvdhoeven, 2021-03-02:
In my setup, bCNC can talk to GRBL over USB even when the rest of my CNC machine is turned off.

I've built a CNC mill with relatively big Nema23 motors (112mm long) which have separate stepper motor drivers. GRBL resides in a "Blue Pill" which I've soldered on a Matrix board with a bunch of wires and TTL buffers to drive the stepper motor drivers. I've got a main power switch which cuts of power to an 1.5kW inverter for the spindle, and to the 36V powersupply for the stepper motors. The "Blue Pill" is then however still powered by the USB cable and bCNC can communicate with it normally. This lets me simulate in bCNC while the machine is nice and quiet.

GRBL is a Free Open Source program, so just grab one of the many compatible uC boards, put GRBL in it and hang that of an USB port.

Harvie, 2021-03-02:
Also you can compile GRBL to run on your PC. But it is far from being ideal:
https://github.com/grbl/grbl-sim