vlachoudis/bCNC #1552: no webcam on Ubuntu

alexlinno, 2021-02-23:
Hi
I run bCNC version 0.9.14-dev(linux py3.8.5) and have a USB endoscope. I tried a standalone python script with a camera cv= cv2.VideoCapture(0) and that works fine.
In bCNC however I have set both alight camera to 0 and web camera to 0 , without any luck. I also tried setting the camera to 1.
So the camera works with opencv but not with bCNC.

alexlinno, 2021-02-23:
O and I cannot remember the version anymore ..but it has worked on older bCNC versions

Harvie, 2021-02-23:
Are there any error messages?

alexlinno, 2021-02-23:
no error , I started to debug it. The camera is created but the icon is always grayed out, Camera start is never called by that might be connected to the  icon/buttons not being enabled

alexlinno, 2021-02-23:
OK I think the problem is PIL which cause Camera.cv to be None. If I manage to fix it, I will close this issue

alexlinno, 2021-02-23:
my cvopen was too new in respect to Pillow. now it works

closed 2021-02-23 by alexlinno

Harvie, 2021-02-23:
Is there anything that can be fixed on our side? eg.: improving dependency requirements described in here:
https://github.com/vlachoudis/bCNC/blob/master/setup.py

alexlinno, 2021-02-23:
No I think after trouble with the camera I manually upgraded opencv and it was version 4.5.1.48.......and that messed things up....Maybe you should just print to the terminal that a module doesn't load. I didn't get anything o the ImportError exception....is that logged?
And of course, thanks for all your efforts!
Alex

Harvie, 2021-02-23:
> Maybe you should just print to the terminal that a module doesn't load.

Can you please try to implement such message and test it on your machine? (you will have to break your setup again :-)

alexlinno, 2021-02-25:
Something like this works:
Camera.py line 19
try:
	from PIL import Image, ImageTk
except ImportError:
	print("Unable to import Image, ImageTk from Pillow\n")
	cv = None

output:
new-config Utils <configparser.ConfigParser object at 0x7fe59dde10d0>
Unable to import Image, ImageTk from Pillow

Harvie, 2021-02-25:
Ok, implemented. Thanks.

commit fef19804 (2021-02-25): Fixes #1552