vlachoudis/bCNC #1563: Sending $$ breaks connection to GRBL 1.1

truglodite, 2021-03-03:
After the latest update to bCNC, I cannot get my GRBL settings with the $$ command in either the bCNC terminal or the CAM "$$" button. I'm running the latest bCNC on updated raspbian. Both python3 and python2 installs are behaving the same. Here is the terminal output when the error happens is consistent:

```
pi@raspberrypi:~ $ python3 -m bCNC
new-config Utils <configparser.ConfigParser object at 0x75d0b930>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 33554522
Not Loading 6 axis displays
Not Loading 6 axis displays
Controller state changed to: Alarm (Running: False)
Controller state changed to: Alarm (Running: False)
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/bCNC/Sender.py", line 787, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/home/pi/.local/lib/python3.7/site-packages/bCNC/controllers/_GenericController.py", line 248, in parseLine
    pat = VARPAT.match(line)
NameError: name 'VARPAT' is not defined
```

...and just to verify:
```
pi@raspberrypi:~ $ python3 --version
Python 3.7.3
```

I don't know python so I am hoping someone else can help sort this one out. bCNC is otherwise working properly for me. I can actually cut parts with it, but having to be careful not to touch "$$" does not inspire confidence.

Harvie, 2021-03-03:
Can you please check if that might be caused by #1544 ?

truglodite, 2021-03-03:
Harvie, that fixed it. Thank you for the info. 

I tried again installing from the latest git (with "float" for $0) it is working. I am guessing this fix has not been implemented for pypi installation methods.

Harvie, 2021-03-03:
> Harvie, that fixed it. 

This would mean you didn't have GRBL 1.1, if this helped you, it means you have grblHAL. How am i supposed to know it?

> I am guessing this fix has not been implemented for pypi installation methods.

Actualy it was included in pypi versions 0.9.14.313 and higher.
Maybe you just need to do `pip install --upgrade bCNC`

truglodite, 2021-03-03:
> it means you have grblHAL

AFAIK I have GRBL 1.1. The ramps board is running a not too old version of grblmega5x. More specifically, this FW:
https://github.com/Enducross/Grbl_MPCNC_Autosquare

I also happen to be working on setting up a GRBL esp32 board. That board is running the latest master of bdring's grblesp32, which also uses 1.1 paramaters AFAIK. The behavior was no different with grblesp32.

> Maybe you just need to do `pip install --upgrade bCNC`

I am likely missing something obvious here. As I said in the OP, this happened with a fresh install of raspbian from the raspberrypi.org image, using either ```pip install --upgrade bCNC``` or ```pip3 install --upgrade bCNC```. Also, after using those methods to install, I did verify the GUI version was 0.9.14.340. So strange, because it's the same version that is working for me now (just installed manually via master zipfile).

beta-tester, 2021-03-05:
i have the same issue with $$ but i am still on python2.7 with my Raspberry Pi.

```
pi@CNC:~ $ sudo pip install --upgrade bCNC
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: bCNC in /usr/local/lib/python2.7/dist-packages (0.9.14.316)
Requirement already satisfied, skipping upgrade: Pillow>=4.0 in /usr/lib/python2.7/dist-packages (from bCNC) (5.4.1)
Requirement already satisfied, skipping upgrade: pyserial; sys_platform != "win32" in /usr/lib/python2.7/dist-packages (from bCNC) (3.4)
Requirement already satisfied, skipping upgrade: numpy>=1.12 in /usr/lib/python2.7/dist-packages (from bCNC) (1.16.2)
```

```
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 29360218
Not Loading 6 axis displays
Not Loading 6 axis displays
Controller state changed to: Idle (Running: False)
```

```
$$
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/Sender.py", line 787, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/usr/local/lib/python2.7/dist-packages/bCNC/controllers/_GenericController.py", line 248, in parseLine
    pat = VARPAT.match(line)
NameError: global name 'VARPAT' is not defined
$0=10
```

```
$I
[VER:1.1h.20190830:]
[OPT:V,15,128]
ok
```

and if i click to "Report" on the File tab, i see this on the console:
```
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bCNC/Utils.py", line 526, in __call__
    return self.func(*args)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/Utils.py", line 690, in sendErrorReport
    ReportDialog(None)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/Utils.py", line 600, in __init__
    from bCNC import __version__, __date__
ImportError: cannot import name __version__

```

Harvie, 2021-03-05:
> NameError: global name 'VARPAT' is not defined

This was fixed, please upgrade pip one more time.
Also i highly reccomend to use latest python 3 to prevent issues. We will soon drop python 2 support completely.

beta-tester, 2021-03-05:
yes, thank you. it is fixed now in bCNC-0.9.14.317.tar.gz (python 2.7) ...

that i installed the python 2.7 version of bCNC was by accident, because it is still the default python/pip version in Raspberry Pi OS. python3/pip3 is installed on the system as well.
(or i tried to use the python3 version at a time where bCNC (python3) was buggy... i can't remember)

i am now on bCNC (python3)
```
pi@CNC:~ $ sudo pip uninstall bCNC
Uninstalling bCNC-0.9.14.317:
  Would remove:
    /usr/local/bin/bCNC
    /usr/local/lib/python2.7/dist-packages/bCNC-0.9.14.317.dist-info/*
    /usr/local/lib/python2.7/dist-packages/bCNC/*
    /usr/local/lib/python2.7/dist-packages/tests/*
Proceed (y/n)? y
  Successfully uninstalled bCNC-0.9.14.317

pi@CNC:~ $ sudo pip3 install --upgrade bCNC
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bCNC
  Using cached https://files.pythonhosted.org/packages/4f/3f/21eca89c78550e30e8937b3e9c27bc8ddbce8cbcda848b4b6b0b3bca68e2/bCNC-0.9.14.317.tar.gz
Requirement already satisfied, skipping upgrade: Pillow>=4.0 in /usr/lib/python3/dist-packages (from bCNC) (5.4.1)
Requirement already satisfied, skipping upgrade: numpy>=1.12 in /usr/lib/python3/dist-packages (from bCNC) (1.16.2)
Requirement already satisfied, skipping upgrade: pyserial in /usr/lib/python3/dist-packages (from bCNC) (3.4)
Building wheels for collected packages: bCNC
  Running setup.py bdist_wheel for bCNC ... done
  Stored in directory: /root/.cache/pip/wheels/22/e6/ba/c8fdfc0e822806e4f68d108dd124f4faff9e89a60182052915
Successfully built bCNC
Installing collected packages: bCNC
Successfully installed bCNC-0.9.14.317

```

thank you for the great software!