vlachoudis/bCNC #1564: autolever bcnc

pp5bk, 2021-03-07:
hello guys,
i builted my cnc arduino/grbl based 3/4 years ago, and i never found a  solution to autolever alarms
MY GUESS , because i tryed all mentioned here on the opened topics its not a bcnc problem,
im using latest 1.1 grbl, latest bcnc
last week i dowloaded ULTIMATECNC , that runs native....and has autolever feature....
and i got the same alarm, autolever starts, sometimes goes further but not touches the metal, 
sometimes its just moves up and down and the alarm stops,
and rare times it does the job...
my guess its a grbl issue ...(yes 1=255)
im not going to describe all ive done trying to fix it, because it will be a long list...
all mentioned here i did.

Harvie, 2021-03-07:
If you want help you will need to exactly describe your hardware setup and configuration, especialy regarding the proble.
also don't use any probing GUI for testing. Read how G38.X g-code works and try to do simple probing by manualy sending these commands:
http://www.linuxcnc.org/docs/2.5/html/gcode/gcode.html#sec:G38-probe

If it does not work as described at that page, it is either bug in GRBL (unlikely) or problem with your HW setup or GRBL configuration.

MARIOBASZ, 2021-03-07:
if sometimes it goes beyond where it should, other times the alarm is activated before reaching the contact level, it must be a hardware problem: bad or false probe contact, or electrical noise interference

pp5bk, 2021-03-08:
Hello Thomas,thank you for reply,
Im using the basics grbl configurations as every body does, i got your message and Mario wrote too on the same direction,
I will get a usb Photo isolator and i  will built some kind of it to put betwen arduino drivers pins,(I did it some time ago) ...maybe not efective.
I will let you know,
Thank you all for reading and answering.
Marcelo

http://www.linuxcnc.org/docs/2.5/html/gcode/gcode.html#sec:G38-probe

pp5bk, 2021-03-08:
Thank you Mario,

pp5bk, 2021-03-09:
hello you all,
i did disassemble the entire electric part, i was using arduino uno(original) tb 6600 drivers(3) switched power suply 24vdv 10A, 3 ak23 15 kg motors bipolar.
so i got other arduino with shield and 3 small drivers a4998, and rebuilt it,
i checked all wiring, one by one from the motors,
i dont use hard limits wiring and switches,
well i guess i found the problem, i dont know why it happens...
when i touch the aligator(+5vdc from A5) on the  machine structure i notice and heard a motor pulse, 
more contacts, more pulses, i could see the x axis moving , y too..very small movements, 
crazy...i shuted down motors power...and atached the aligator on the machine, and  started to search for the 5v continuity on the motors wires, one by one and couldn find it on the other side (arduino and drivers) it only occurs with power applied to them,
of course g38.2 is always triggered, intermittent on it...sometimes it worked out...
i noticed it i guess because i  did set a different step 1/4(for testing) on this hardware, but the thing is...the pulses generated by the aligator was doing the mess, intermittent!
i will use a usb photo coupler, and will put photo couplers (4n25) on the digital pwm pins too, the arduino will be alone, 
sorry for my bad writing english.
i hope it will bring some attention on this issue,
im very concerned about xing ling quality motors...

thank you all
marcelo