vlachoudis/bCNC #1576: Random HOLD triggers

probe3, 2021-03-21:
I've been running bCNC for about 2 years and haven't updated it on my Windows10 laptop. I have the latest Fusion360. Lately I've been getting random HOLD triggers during my programs. I have to babysit the machine because I have to manually resume the operation. During a simple ramped contour operation it triggered HOLD about 20 times. I tried to an adaptive operation and it kept happening so often that I shut the whole thing down.

probe3, 2021-03-21:
(FIXED) - Replaced USB cable from laptop to controller.

closed 2021-03-21 by probe3

Harvie, 2021-03-23:
replacing USB cable might help, but i guess you should step up your shielding game anyway... especialy cables to motors and switches.