vlachoudis/bCNC #1582: PCB milling problem

ours-polaire, 2021-04-11:
Hello,
I haven't been able to solve a problem with cutting a board for 2 weeks. I am working on a project and I went on the road - Kicad - FlatCAM - bCNC. The problem arises at the last stage after auto-leveling. First I worked with CNC Shield v.3. There, the parameters on the Z-axis changed arbitrarily. I changed the board with Annoy tools - 2208 drivers. Everything is fine on the Z axis, but now there is a problem on the Y axis. I noticed that where there is a red zone at the auto-level (higher level of the board) the tracks are cut on top of each other. I tried OpenCNCpilot under Windows and there is the same problem. I repeated the FlatCAM files, but nothing changed.

KiCad - version 5.1.7.
FlatCAM - 8,991 and 8,994.
bCNC - 0.9.14.52

Please direct me where the problem is so that I don't have to print the topology and use old tried and tested methods with an iron.

I put a zip file of G-code, created on FlatCAM:
[big_roaster_v4.zip](https://github.com/vlachoudis/bCNC/files/6293002/big_roaster_v4.zip)

Thanks!

SonOfSardaar, 2021-10-03:
@ours-polaire this sounds like a problem in your stepper motor setup. check the following things
1. Your motor driver (2208 driver or which ever you are using) has enough current tuned so that motor has enough power.
2. Speed and acceleration values for your x and y axis are not too high that your motors are not able to cope (step missing)

these things could cause axis to move incorrectly and give bad results

ours-polaire, 2021-10-04:
Thanks,You advise me very precisely. It turned out that I forgot to adjust the driver current.