vlachoudis/bCNC #1585: GUI sizing - 7" Touchscreen

rjd1234, 2021-05-11:
I'm working on a DIY router with a 7 in Touchscreen and Raspberry Pi with Protoneer hat. I'm having an issue with the GUI not fitting on the screen... the bottom parts are cut off.  I have played with screen resolution and tried collapsing certain sections of the GUI, but I still can't figure out a way to get to the bottom.

Is there a way to resize the fonts?  Maybe that would make everything a little more condensed.

Is there a way to move sections around?  I really don't need the large image showing the spindle position on the right... I could replace that with the jog buttons.

[image]

Thanks!

Harvie, 2021-05-11:
Why don't you switch to fullscreen mode?

rjd1234, 2021-05-11:
> Why don't you switch to fullscreen mode?

Hi Harvie, I tried both expanding the screen (ie clicking on the square and arrow in the upper-right corner).  This did not make a difference.  Is there another way to set it to fullscreen?

The picture might be misleading because it looks like there is a black area that is being unused... That is just how the bezel and panel look in the picture... all of LCD portion of the screen is being used.

Harvie, 2021-05-11:
It might be specific to desktop environment you use on linux.
Perhaps right click window title or window button on bar? Maybe there is some keyboard shprtcut.
According to swift googling raspbian uses LXDE and that uses  Alt + F11 to toggle fullscreen. But i am sure it can be configured to something else.

rjd1234, 2021-05-20:
Hi,

The issue is not making getting it to go fullscreen.  Here is an image showing fullscreen.  You can see that the black area around the outside is a bezel.
[image]

I have played with the resolution. I have resized everything I can think of.  I have collapsed certain portions, which helps, but I still can't get to all parts of the GUI.  I really want to use bCNC, but I need to modify the GUI somehow.   

Couple of really simple ideas:
1.  On the Control tab, make the Status / wpos / mpos (basically everything above the State portion) collapsable.
2. Move the "Status Idle" line to the empty space in the upper right portion (under the Terminal) tab 
3. Allow me to adjust the font size of the control panel

Anything would help.  Please let me know if it is possible to adjust the GUI somehow.

Thanks!

karoria, 2021-05-20:
Hi @rjd1234,
I am also interested in getting a GUI which suits to raspberry pi with 7 inch touch display, and I am sure there will be lot of people like us. However, I am not a software guy. If you get to a good solution, kindly share your project if you don't mind.
Best,
Ravi

Harvie, 2021-05-20:
@rjd1234 you are not in fullscreen mode. if you switch it to fullscreen, you will be able to use FULL area of screen for bCNC, namely reclaiming the highlighted sections currently used by window manager:

[image]

rjd1234, 2021-05-25:
Harvie,

Like I said earlier, I toggled between fullscreen and tried everything.  Fullscreen is not the issue.  Here is an image in fullscreen.  I have also minimized the section on the bottom (even though that isn't in this picture... trust me, I did), but that didn't give me access to everything either.  

[image]

humneraudio, 2021-06-15:
I think the other issue is the screen resolution. It looks like you have the raspberry pi 7" monitor which is only 800x480

I also run a raspberry pi - Here is mine on a waveshare 7" - but its res is 1024×600

If I hide the State section, I can see the entire Control section. But for me I don't use the mouse controls.

Then if you want to undecorate + full screen bCNC automatically on opening, you can add this to your lxde-pi-rc file under pi\.config\openbox

<applications><application class="Toplevel"><maximized>true</maximized><decor>no</decor></application></applications>

"Toplevel" is the reference to the main bCNC window

[image]

MarSik, 2021-07-06:
I have the same issue, even though I was planning to re-purpose an old Asus EEE 701 (800 x 480 px).

I wonder if Tkinter supports scrolling of the controls column or if a different layout could be used for such small displays. It has been a while since I used it in a project.

rjd1234, 2021-12-07:
To close the loop on this... First I reformatted the buttons and moved things around and that worked, but I never liked it.  I ended up ordering a 10.1" screen from Aliexpress that was cheaper and has 1280x800 resolution and everything is great.  I got the touchscreen version, but I tend to use the mouse more anyway.

[image]