vlachoudis/bCNC #1587: Camera orientation problem

CMunayco1, 2021-05-20:
Hi everyone! I'm trying to get some holes drilled in my pcb with my CNC3018 and an small endoscope. I'm using the toner transfer method and I want to have the holes aligned to the pcb orientation. I'm following this tutorial https://github.com/vlachoudis/bCNC/wiki/Orientation-of-gcode, however I can't get an angle as shown there. I get this angle value instead: 
[image]
I tried drilling but the holes are not aligned. Maybe many things have changed since the tutorial was written, but I can't get it to work for now. Is there any way to work around it to get the holes aligned to the orientation? I hope you can help me with this obstacle. Thanks in advance!

CMunayco1, 2021-05-21:
I managed to do it. I'll be doing some more testing to let you know the results.

drewnabobber, 2021-05-22:
Your error is quite large.  Might need to calibrate steps/mm in your controller.  The 3018 is quite poor out of the box.

engrjawadulhassan, 2022-01-03:
Hello,
bCNC is my daily use software for my CNC based on GRBL (Windows 10 x64), everything works great, Nowadays i am struggling with orientation of already etched PCB for drilling without camera.

I am having the issue while drilling after completing all the steps mentioned here on GitHUB in this issue
https://github.com/vlachoudis/bCNC/issues/140#issuecomment-179965895

After reading some more issues/threads , I recently re-calibrated my CNC for accuracy.

But the issues i am facing is that , every time , after adding the points as per mentioned in above link , and clicking on orient button , visually in software (bCNC) pane , it oriented the PCB, but after this when i try to process the drill job , it gives error while moving along x-axis (this error is about 0.4mm at first drilling point) , If i remove this error at home point by jogging the machine manually the error get removed at first point of drill. But after some of accurate drill points the error again increasing with the drill points increasing.

Please help me regarding this , if i am doing something wrong, or missing anything.

Thanks

engrjawadulhassan, 2023-06-14:
Hello,

I previously encountered an issue pertaining to camera orientation. However, I've discovered an online tool named 'Pcbdrillutility' that has proven effective in resolving this issue. The tool, as discussed in this Hackaday post (https://hackaday.com/2013/08/04/drilling-pcbs-with-cameras-and-math/), is written in Python and is openly accessible on GitHub (https://github.com/nextime/pcbdrill).

The operational capabilities of the tool can be seen in this video: https://www.youtube.com/watch?v=tzI445XsPDs.

I believe that a detailed review of this Python code could provide valuable insights to enhance the accuracy and efficiency of the orientation function in bCNC. I kindly request your assistance in investigating this possibility.
Thanks.