vlachoudis/bCNC #1588: New Windows isntalation?

pedjas, 2021-05-23:
The latest Windows instalation in releases is from 21 Nov 2018.
It would be good to have updated one.

Harvie, 2021-05-24:
use pip install.

closed 2021-05-24 by Harvie

pedjas, 2021-05-24:
What is pip?

Harvie, 2021-05-25:
Please read documentation: https://github.com/vlachoudis/bCNC/wiki/Installation

pedjas, 2021-05-25:
Ok, I see. To be user of bCNC I first have to get a degree in Python programming.

MARIOBASZ, 2021-05-25:
Jajaja. 
Courage, it is true, it is not automatic, but bCNC's performance is really very good.

Harvie, 2021-05-25:
> To be user of bCNC I first have to get a degree in Python programming.

To install bCNC on windows you need to install python, and perhaps on some private school this alone might get you a degree in programming.
Also you need to install windows first, which does not make you CEO of microsoft :-)