vlachoudis/bCNC #1590: How can I know where I have to plug a laser on a CNC card

Paugeman09, 2021-05-26:


Paugeman09, 2021-05-26:
[image]

[image]
[image]
[image]

carl1961, 2021-05-26:
Most connect to spindle pin witch is usually pin D12 Read up on your laser module, it may need a driver module
[image]

check your    cpu_map_h

https://craftweeks.com/en/how-to-steup-grbl-for-laser-engravers/

rjd1234, 2021-05-27:
I typically generate toolpaths in Fusion 360 and the Laser cutting post processors that I have used activate the laser using S1000 and then M3 and M5... This is the same way that a spindle would be turned on it off. 

So, yes. Connect it to the spindle terminals.

Paugeman09, 2021-05-27:
So if I understood I just have to plug it in the pin D12, but my laser have three ports, which one should I plug ?
What do you call a driver Modul ? 
Thank you for your help

Paugeman09, 2021-05-27:
> I typically generate toolpaths in Fusion 360 and the Laser cutting post processors that I have used activate the laser using S1000 and then M3 and M5... This is the same way that a spindle would be turned on it off. 
> 
> 
> 
> So, yes. Connect it to the spindle terminals.

What do you call a spindle terminal ?

Paugeman09, 2021-05-27:
> Most connect to spindle pin which is usually pin D12 
> 
> Read up on your laser module, it may need a driver module, 
> 
> 
> 
> check your    cpu_map_h
> 
> 
> 
> 
> 
> https://craftweeks.com/en/how-to-steup-grbl-for-laser-engravers/

So if I understood I just have to plug it in the pin D12, but my laser have three ports, which one should I plug ?
What do you call a driver Modul ? 
Thank you for your help

Paugeman09, 2021-05-27:
> Two pins are for power, you need to connect to suitable power supply according to laser specifications.
> 
> Third pin is on/off PWM signal that can be provided by your controller board directly.

Thank you very much for your help, I just have a last question, how can I know which pin is the power pin ?

Harvie, 2021-05-27:
Why don't you ask manufacturer/distributor of that laser?
I would expect black/red/yellow to be gnd/12V/signal. BUT IF I AM WRONG, YOU WILL DAMAGE THE LASER PERMANENTLY.
So ask your vendor for manual/datasheet.

Harvie, 2021-05-27:
Also this laser is powerfull enough to permanently damage your vision just by short glimpse of reflection from little piece of reflective surface. Also can set your house on fire if left unsupervised. And you need proper ventilation ducts to suck the smoke from cutting, fumes can be toxic.

Paugeman09, 2021-05-27:
> Why don't you ask manufacturer/distributor of that laser?
> 
> I would expect black/red/yellow to be gnd/12V/signal. BUT IF I AM WRONG, YOU WILL DAMAGE THE LASER PERMANENTLY.
> 
> So ask your vendor for manual/datasheet.

I will try to contact him thank you I hope he respond

Paugeman09, 2021-05-27:
> Also this laser is powerfull enough to permanently damage your vision just by short glimpse of reflection from little piece of reflective surface. You need at least wavelength specific glasses, but more likely fully enclosed machine. Also can set your house on fire if left unsupervised. And you need proper ventilation ducts to suck the smoke from cutting, fumes can be toxic.

Thank you for the prevention I had buy some protection glasses