vlachoudis/bCNC #1593: Draw vector graphics inside bCNC

eyalmai, 2021-06-03:
Hello everyone,

Is there any way to draw vector graphics on the canvas of bCNC ?

I mean basic lines and arcs, that can later be used to generate gCodes from.

Instead of importing DXG/SVG to the canvas from InkScape -> tools to draw directly inside bCNC ?

Thanks !

Harvie, 2021-06-03:
Try "Probe->Record", it can produce some elementary graphics and "CAM->CAM->Generator".
It is not really meant to produce whole design. But  "Probe->Record" allows you to jog to points and then for example cut line between them. Also you don't need to jog, you might just enter coordinates to current workspace.

Fully interactive CAD would be nice, but we absolutely have to rewrite canvas from scratch, so i don't think it is good idea to add it right now. Also truth is that we don't know when the canvas will be rewritten, it's problem for years, but no solution is coming so far. but i would definitely postpone adding new features to canvas until we fix it.

DodoLaSaumure, 2021-06-03:
Hi @eyalmai 
Try using CAM / Generator / Simple Line, simple Arc, Simple rotate, simple translate etc..
Not because I wrote those plugins, but because I find them very useful, and friends of mine reported their interest for those tools.
Have a good day
PIerre

[image]

DodoLaSaumure, 2021-06-03:
Please refer to #1333 to see screenshots and use cases.
All the best
Pierre

eyalmai, 2021-06-04:
> Try "Probe->Record", it can produce some elementary graphics and "CAM->CAM->Generator".
> It is not really meant to produce whole design. But  "Probe->Record" allows you to jog to points and then for [...]

Beatiful, thank you !!
Really liked the idea of probing and then generating gcode, it opens a new world !

interactive cad sketcher on the canvas , using the mouse and dimensioning/constraint would be a really powerful, where is there info about the limitations of the current canvas ?

many thanks Harvie !

eyalmai, 2021-06-04:
> Try using CAM / Generator / Simple Line, simple Arc, Simple rotate, simple translate etc..
> Not because I wrote those plugins, but because I find them very useful, and friends of mine reported their interest for those tools.

Those are really useful Dodo, i tried them and it seems to work good for me.

Many thanks !

Do you believe theres a chance to use the mouse cursor location instead of the numeric inputs, to define the different geometries ? 

i understand from Harvie that the way to complete CAD is not straight forward due to limitations of the canvas, but maybe such a simplistic CAD ability is a nice start ?

Harvie, 2021-06-04:
> where is there info about the limitations of the current canvas ?

https://github.com/vlachoudis/bCNC/issues/591

TLDR; it is written in python and therefore too slow. We need native (C or Rust) library that takes g-code and visualises it while enabling for interaction and editing and then call that from python. Otherwise it will be huge pain, especialy when working with huge files.

DodoLaSaumure, 2021-06-04:
> Do you believe theres a chance to use the mouse cursor location instead of the numeric inputs, to define the different geometries ?

I do believe that this would be great if we could had a CAD inside bCNC, but HMI is already very heavy for users, and difficult to code for developers.
Everything is possible, but I am afraid that it would be a painful dev, both for users and coders.
...
That's the reason why I used flexible plugins to manage this use case, I think they cover most basic usages.
Have a great day
Pierre