vlachoudis/bCNC #1594: OS native file selection dialog

gege2b, 2021-06-04:
Hi everyone
bCNC is a very good software, and even if far from perfect, it work very well for me, thank you for this !

I have a question (and if answer is "no", it could became a feature request :) 

I'm often using bCNC on a Windows 10 with itinerant profile (hence, it's a \\network\path\to\Documents) and bCNC's file selection dialog can't access these path (with some another annoying things)

So here's my question : is there a way to use another (or OS native) file selection dialog ?

I've searched the issues tracker but found nothing

Thank you for your time
Gege

Harvie, 2021-06-04:
> So here's my question : is there a way to use another (or OS native) file selection dialog ?

If you can figure out how to do this in python/tkinter without sacrificing portability or features, i will accept your PR.

gege2b, 2021-06-30:
Hi,
I understand... And if I was good enough in python and tcl/tk, I certainly would've tried to do something but actually I can't even figure out where to look at

Actually, I found that doing a copy/paste of the path in the file name input box seems to work, but I don't know if it access the actual network drive or the local "offline" copy