vlachoudis/bCNC #1596: BCNC and PLC Mduino42+:Not connected

yvesbel, 2021-06-13:
Hi,
     
I have a problem to connect PLC industrialshields Mduino 42+ and BCNC (it seemed a good idea to do that ,before I had only an Arduino mega without the package found inside the Mduino42+ ( separed power supply,optocouplers,etc) .You can see on below pictures the connection and the messages in return:

[image]
[image]
I can read the arduino mega card number but it's impossible to connect the PLC and BCNC together ("Not connected" message )as you can see
The compiling process with Arduino IDE (1.8.15 version) done before is good with no errors messages (Librairies for Mduino installed, file Cpu_map modified according with Mduino42+ environnment).
My version for Python is 3.9.15 ,for Pip is pip-21.1.2,for Grbl is grbl-Mega-1.1f .
Could you help me ,I think it'is a good case to solve (PLC Industrialshields provide an industriel environnement around Arduino products ) but for the moment no issue to pilot my DIY CNC.
Thanks per advance

Harvie, 2021-06-14:
Hello, i am not sure about grbl-Mega compatibility. Technicaly it is not a GRBL and there might be slight variations in communication. Can you please try with original GRBL to see if there is a difference? If there is some issue in grbl-Mega compatibility we might try to fix it in bCNC. They might even want to fix that in grbl-Mega.

**TBH i would strongly reccomend you to use grblHal for new builds.** But i am not sure if that can be used with your hardware, maybe you can replace the "arduino" board in your PLC box for something more recent (32b), while keeping the isolators and stuff. There are boards with 32b microcontrollers with same pinout as arduino boards, so grblHal can be used on that. Boards are very reasonably priced (comparable to arduinos). 8bit grbl is mostly dead, so we really should move on.

Harvie, 2021-06-14:
If your plc has actual "Arduino Mega" board in it, you can just swap it for "Arduino Due", which will run grblHAL.

yvesbel, 2021-06-15:
Hi,
Thanks for your answer.
I have found last morning the solution of my problem.In fact ,on the PLC industrialshieds Mduino (you have to read with a lot of thoroughness the userguide ,see website on this spanish company) ,you have a button to put on « OFF » after the flashing process and it was the BUG !!I will send you a global return after my first tries on my DIY CNC.
Thanks you.
See you later.

closed 2021-06-15 by yvesbel

yvesbel, 2021-06-15:
Hi,
As I said,"VOILA "my return after my last tries and hardware connections:all is OK now! 
Last step to see:6 hours of uninterrupted production without alarms.Let's cross our fingers...

JordanBln, 2021-09-15:
Hi Yves,
I hope your production worked well. 
I want to do the same thing as you but when I change pins in Cpu_map, I've "Q_X is not declared in this scope". What did you do to solve it ?
I thank you in advance.

yvesbel, 2021-09-15:
Hi Jordan,

My production works well and I am building a web site on Shopify to sell my products in wood,thanks.

You have 2 separate loops inside the bnc software :The first loop is the appropriate here. See below my new cpu_map,check all lines in yellow ( you have to change allocations according with the interrupt vector/pinout diagram).I don’t know your level and knowledge on Bnc software and hardware but I think it would be a good step to begin to solve your problem.

Regards

Yves