vlachoudis/bCNC #1597: Skipping a line of gcode then stopping at "idle" right before the next Z move

humneraudio, 2021-06-14:
I have had this issue a few times now. I can confirm its not the gcode as I can run the same program through its entirety and its fine.

What happens is the machine will skip a line of gcode and just move to the next coords. It will then carry on from there until it reaches the next Z movement and then stops at "idle".

If I try and press RUN nothing happens. 

Is this normal behavior? My thought maybe that bCNC is detecting that a line is "skipped" or doesn't reach the grbl controller and carries on until it reaches the next Z movement? 

Unfortunately each time its happened I haven't checked on the terminal to see what was going on. I tried running the same job but I wasn't able to replicate the problem.

I'm running 9.14 on a raspberry pi4

closed 2021-07-13 by humneraudio

Harvie, 2021-07-14:
hello @humneraudio , have you figured this out?

humneraudio, 2021-07-14:
No I haven't. I haven't been able to replicate it again. I've run plenty of "dry runs" of the same gcode files to try and make it occur again, but all the jobs finish fine.

I decided to close this as I'm about to move onto a completely new setup, I'm moving onto grblHAL on a completely fresh Linux install(and different computer) and also running it over Ethernet (which works really well so far).

I thought no point looking into it further now that everything will have changed. I'll see how this goes. If I end up running into the issue again, hopefully I can get more info when it happens.

Harvie, 2021-07-14:
Sounds like noisy USB/Serial wiring to me.
grblHAL is great. there is some minimal ethernet/ip support in bCNC, but i didnt upgraded my machine to 32b microcontroller yet, so haven't really tried it.

humneraudio, 2021-07-14:
Yeah, an issue on the serial line is the only explanation for it to skip a line really. I think I was more wondering about the behavior after it skipped the line.

Anyway.

I've been testing the grblHAL breakout board on Teensy 4.1 with Ethernet - https://github.com/phil-barrett/grblHAL-teensy-4.x

The ethernet connects without any issue on the latest bCNC version with socket://192.168.60.200:23 (that's the IP of my teensy's ethernet port). Running on an old PC(on latest Lubuntu distro) with dual NIC so the controller gets it own NIC away from my normal network.

I haven't put much time into yet, but so far its been stable.