vlachoudis/bCNC #1600: Protect machine table

MARIOBASZ, 2021-06-29:
A good safety measure would be to set a minimum height for the tool in use.
If the height of the Z probe relative to the machine base is constant, a routine could be implemented that will interrupt the work if the Z height is too low and reports in the "height too low" style.

ZASto, 2021-07-19:
A good safty measure is to reference your Z axis when doing cutouts to machine bed.

MARIOBASZ, 2021-07-19:
@ZASto , My reasoning is to protect from human errors (I make too many ...)
Suppose I reference Z for base = -20
By mistake, I write
G0 Z-24
Crash, I broke another end mill

ZASto, 2021-08-28:
Why would you enter negative Z values if you are referenced to machine bed?
Any negative Z will end in F"#$%& bed.

MARIOBASZ, 2021-08-28:
The Gcode could have a lower value than the bed, or I am reasoning wrong.

ZASto, 2021-08-28:
For any axis, g-code can have a value between -∞ to +∞.
If you write g-code manually, you can always plow through your bed.
Z referencing is something that you have to figure out yourself and to be consistent.
When I play with 2 1/2 D reliefs in wood, Relief is referenced to the top of the material.
For cutout, tool is referenced to the machine bed so that Z0 will not damage the bed (besides I'm using a sacrificial board on machine bed, just in case).
For generating g-code I'm using Vectric Aspire. Couple of clicks more saves the machine bed.