vlachoudis/bCNC #1602: Support of Tool state for grblHAL firmware

kfmut, 2021-07-04:
Hi,

I'm doing some test with grblHAL firmware particularly for native tool changes and noticed that bCNC isn't following tool change protocol for `Tool` state( https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol ). If the tool change policy is set to `Send M6 commands` bCNC doesn't suspend streaming of g-codes from a file, which results in error 40. Are there any plans for fully implementing the tool change protocol of grblHAL?

Harvie, 2021-07-07:
Didn't knew about the protocol. Pull requests are greatly welcome.

kfmut, 2021-07-07:
@Harvie thank you for clarification!

> Pull requests are greatly welcome.

Yes, there are! :-D

kfmut, 2021-07-08:
@Harvie

Sorry for disturbing you again :-) I was browsing through issues/discussions in milestones and those labeled as “research and code cleanup”, and noticed few about bCNC future. So I’m curious was there any outcome of those discussions? What is the future like for bCNC? Are there any noticeable forks of bCNC active now or partial rewrite attempts?

Harvie, 2021-07-09:
AFAIK There are no plans for "hard forks" as we are open for any reasonable improvements. There are development forks of individual contributors. The good stuff usualy pops up in PR list here in this repository to be reviewd and merged. If you ask me, the major pain is Tkinter and especialy unaccelerated canvas. We need python module written in C that will take g-code blocks and render them. Allowing them to be selected and stuff, letting us to freely rotate the view, do antialiasing and stuff. Python code is just several orders of magnitude too slow for this.

kfmut, 2021-07-09:
@Harvie

Thank you!