vlachoudis/bCNC #1603: Gcode Rendering Issue & Auto Leveling Issue

PaladinEnabling-Technologies, 2021-07-18:
Hello There, I've been using bCNC for a little while now to make PCBs and it is great. Recently though, I started seeing some strange behavior with the auto leveling. What seems to happen is that the probe triggers prior to actually touching the copper section:

[image]

The number 1 picture shows where the probe is triggering the auto level without actually touching and sometimes the probe does travel all the way to touch the copper to get the auto level point. The tip on the copper is the light reflecting off the V bit, if the light reflection tip and the v-bit tip touches, then you know that the v-bit is making contact with the copper.

This can also be seen on the auto leveling data map where I don't have a ~2 mm fluctuation in the blank pcb copper layer:

[image]

The spindle has been disconnected to try and rule out any 

Another follow-up question I had was related to the rendering. When uploading larger G-code files, then the job visualization area is not populated with the entire G-code file:

[image]

A notification did appear once stipulating the following: "Rendering takes TOO Long. Interrupted..."

Example of the how the visualization should look:

[image]

I'm not really that worried about the visualization of the G-code file as the G-code is still executed completely even though the visualization is not complete.

The main concern is related to the auto leveling, looking for some insights here to try and resolve it. Thank you and looking forward to feedback on the matter.

PaladinEnabling-Technologies, 2021-07-22:
Hello Everyone, 

I managed to resolve the issues that I logged here. 

**Issue 1 - Z-probe triggering prior to it actually touching the copper plate:**
**********************************************************************************
My spindle ground was not sufficient and would sometimes incorrectly trigger. I had added an additional grounding pin which is linked to the same grounding pin I attach to the spindle.

This additional grounding pin resolved the probing issue that I had so if you are experiencing this, then please try this.

**Issue 2 - Rendering not completing in the job viewer window:**
*********************************************************************
There is a timeout function which impacts the rendering which I was not aware of. Setting this setting higher allows for the job to render completely.

[image]

closed 2021-07-22 by PaladinEnabling-Technologies

Harvie, 2021-07-22:
> There is a timeout function which impacts the rendering which I was not aware of. 

Yes, this is problem. Python is extremely slow. We need to write OpenGL based python module in C language, compile it to native library and call that from python. Unfortunately it is not that easy. Perhaps we can use some library like coin3d (this is what FreeCAD uses for this exact reason). But it's not easy problem to solve, i've been looking for solution for some time already...