vlachoudis/bCNC #1608: What is its communication protocol？

Waterman178, 2021-08-04:
I put grbl into the blue pill . I want it to communicate with bcnc, but I don't know the  protocol. After I analyzed the protocol with virtual com and debugging assistant, I found that it has been sending 0x3f. Although I successfully opened com, it prompts that it is not connected. If I casually reply to some data, it will prompt that a connection has been established, but the title is still not connected。How can I know the detailed communication protocol？
[image]
[image]
[image]

Harvie, 2021-08-04:
`0x3F` is question mark `?`. This is command which bCNC uses to request status report from GRBL.
What grbl did you put in your blue pill?

I reccomend using grblHAL: https://github.com/grblHAL/core
It has support for blue pill: https://github.com/grblHAL/drivers

Basic GRBL protocol is described here: https://github.com/gnea/grbl/wiki
and grblHal introduces extensions to basic grbl protocol described here: https://github.com/grblHAL/core/wiki

Harvie, 2021-08-04:
If you have issues connecting grblHAL with bCNC, please see the README:

https://github.com/vlachoudis/bCNC/blob/master/README.md
You can find there paragraph with following title: _IMPORTANT! Motion controller (grblHAL) settings_

Also you might want to check the following https://github.com/grblHAL/core/wiki/Compatibility-level

closed 2021-08-04 by Harvie