vlachoudis/bCNC #1611: User Buttons and Database entries are not saved

Baeckchen, 2021-08-20:
Ich kann die User Control Button ändern aber nach einiger Zeit sind die Einstellungen wieder weg und es wird der Standart angezeigt.
Ebenfalls die Database Eintrage für Material und EndMill kann ich ändern und neue anlegen diese werden aber nicht gespeichert. 

I can change the user control button, but after a while the settings are gone and the standard is displayed. I can also change the database entries for Material and EndMill and create new ones, but they are not saved. 
System is Raspbian GNU/Linux 10 (buster)
bCnC Version 0.9.14

humneraudio, 2021-08-21:
I noticed I have a similar issue when adding in new shortcuts, after I close bCNC and open it again, they don't save sometimes.

I believe the issue is that you need to close bCNC but WAIT for at least 30 seconds before opening bCNC again, I think there is a process running after bCNC closes to save new config to the user side config file and it doesn't happen instantly. If you open bCNC too quickly after closing it I think the process halts and doesn't complete.

Give it a go and see what happens.

Baeckchen, 2021-08-22:
Ich werde das Testn
Aber warum werden die Datenbankeinträge für Material and End-Mill nicht gespeichert

OK I'll test it.
But why are the database entries for ccc not saved 

Andreas

Baeckchen, 2021-08-22:
Ich habe immer mindestens eine Minute bis zum nächsten Start gewartet aber nach dem 4. oder 5. Start sind wieder allen Einträge weg.

I always waited at least a minute until the next start, but after the 4th or 5th start all entries are gone again. 

:-(

treki1, 2021-09-08:
In welchem File werden die gespeichert? Auch im .bCNC config File?

Baeckchen, 2021-09-13:
Ich habe keine Ahnung wo die Datei gespeichert wird und wie sie heist. Daher habe ich nach Dateien gesucht die neu sind oder geändert nach dem gestrigen Datum aber nichts gefunden.

I did not know where the file is saved and what the name is.
I`m looking for new and edited files at this day, but I can´t find anything

Baeckchen, 2021-10-02:
Ich habe die .bcnc Datei gefunden im Verzeichnis /home/pi/ und alle Einträge wie z.B. der User buttons sind da. Kann es sein das die Datei bei einem update des Systems überschrieben wird ?

I´ve found the .bcnc file, user button are stored there. Can it be that the file is overwritten when the system is updated?

ltspicer, 2021-10-03:
Da ich mit Linux arbeite, kann ich das nicht sagen.
Aber das ist eher unwarscheinlich.

Baeckchen, 2021-10-03:
Danke für den Tip, werde mir mal ein zweites Speichermedium für die gcode Datein installieren.

Thanks for the tip, I'll try to install a second storage medium for the gcode files.