vlachoudis/bCNC #1614: autoleveling and manual tool change tutorial german

treki1, 2021-09-07:
I wrote a german tutorial for autoleveling and manual tool change.
If you want, you can transfer it to the wiki.
I hope I have made a useful contribution to bCNC.
[bcnc_autoleveling_anleitung.pdf](https://github.com/vlachoudis/bCNC/files/7123569/bcnc_autoleveling_anleitung.pdf)

Harvie, 2021-09-08:
@treki1 thank you, you should have write access to the wiki, so you can add it yourself if you wish...

treki1, 2021-09-08:
How can I upload pictures to include in the wiki?

treki1, 2021-09-08:
Uploads are disabled :(

[image]

Harvie, 2021-09-08:
Yes, you have to do the same thing as with the translation. Unfortunately :-(
I don't know why github does not allow to upload images directly to the wiki...

https://github.com/treki1/bCNC/tree/doc

treki1, 2021-09-08:
Done ^^
https://github.com/vlachoudis/bCNC/wiki/Autoleveling-und-manueller-Werkzeugwechsel-mit-bCNC

Harvie, 2021-09-09:
please also make pull request from https://github.com/treki1/bCNC/tree/doc to https://github.com/vlachoudis/bCNC/tree/doc
Otherwise the files will disappear when you update your fork to reflect the upstream.

ltspicer, 2021-09-09:
Is this ok?
https://github.com/vlachoudis/bCNC/pull/1617

ltspicer, 2021-09-24:
Why is the pdf version deleted?
https://github.com/vlachoudis/bCNC/wiki/Autoleveling-und-manueller-Werkzeugwechsel-mit-bCNC
At the bottom of the document.

ImranKhan203, 2021-09-28:
Hello,

Please help me out.
I am using GRBL 0.9 with the help of bCNC software and I want to attach Tower Lamp/Stack Light on our CNC machine.
I would like to add 2 outputs on our machine to indicate if we’re running the program (green light) or if we’re in pause (called ‘hold’ in the program / red light). For that, I have to work on bCNC that is the software used in all our machines.

So, when the program is running, the green light has to be lighted, and if we’re in pause, the red light has to be lighted. The orange one will probably be used for something else in the future.
To do that, I’ll have to detect the state of the program and send it to the lights (2 outputs).

Please tell me in which file of bCNC and which type of program I have to use?

Here is the file which I am using on Python 2.7.16 :
[bCNC01.ZIP](https://github.com/vlachoudis/bCNC/files/7245704/bCNC01.ZIP)

I want to attach this type of Tower Lamp/Stack Light :
[image]

Please suggest me in which file of bCNC and which type of program I have to use?

ltspicer, 2021-09-28:
Pls create a new (issue) post.