vlachoudis/bCNC #1615: Edited german translation

treki1, 2021-09-07:
I edited the german language.
If you want, you can update these two new files (mo and po). 
[LC_MESSAGES.zip](https://github.com/vlachoudis/bCNC/files/7124722/LC_MESSAGES.zip)

Harvie, 2021-09-08:
Please make pull request if you wish these changes to be included in bCNC.

treki1, 2021-09-08:
Done.
I have posted it here: https://github.com/vlachoudis/bCNC/pull/1092

treki1, 2021-09-08:
Hmm I don't know how to fork. I do not understand :(
Do you have a tutorial for this (maybe even in German)?

Harvie, 2021-09-08:
You already have bCNC fork under your account: https://github.com/treki1/bCNC
Just commit the changes to your fork and then create pullrequest from your fork.

treki1, 2021-09-08:
Ok done :)
Thx for support

closed 2021-09-09 by treki1

Harold-Fox, 2023-03-14:
From the very first installation bCNC has appeared partly translated. Within the last days I checked the bCNC.po file and saw that everything is correctly done. Why do not all translations show up on the screen?
/home/neuland/Dokumente/Linux CNC/bCNC/bcncpo-translation.png
/home/neuland/Dokumente/Linux CNC/bCNC/error-report.png
Above is an example of the report window and its translation within bCNC.po

One more question: Where do the numbers after the file name come from? I cannot find a hint to them in the related file. Example:
#: bCNC/ToolsPage.py:1264 bCNC/EditorPage.py:48 bCNC/plugins/box.py:317
msgid "Cut"
msgstr "Schneiden"