vlachoudis/bCNC #1618: I can't open with bcnc.bat

MARIOBASZ, 2021-09-09:
I have the following problem: On a new pc I installed python 3 and downloaded bCNC. I open bCNC from command line with
py -m bCNC
However, if I download a new version to my computer (on my desktop) and I want to open from
bcnc.bat, I can't open it.
Path error? On my old laptop I don't have that problem and I can try modifications, new add-ons, etc.
  Thank you very much if anyone can help me! I guess it must be a silly mistake.

Harvie, 2021-09-09:
This is no longer supported method of launching bCNC and will be probably completely removed in future.

closed 2021-09-09 by Harvie

MARIOBASZ, 2021-09-09:
Hi Harvie, thanks for replying.
Today I can't have two versions of bcnc on the same machine?
How should I proceed to add a plugin or change a file?

Harvie, 2021-09-09:
If you run `python -m bCNC` in the development directory, it does start local copy of bCNC instead of system-wide.
It is described here:

https://github.com/vlachoudis/bCNC/wiki/Installation#from-git