vlachoudis/bCNC #1620: Advanced halt functionality [Feature request]

geobruce, 2021-09-10:
Sometimes there is a situation where I'd like to pause the running g-code especially for longer jobs. To check if the end-mill is still inserted deep enough in the spindle and it's not vibrating loose.

And if it happens to be loose and needs to be reinserted I'd like to go to the predefined point to measure the z-origin.

For now I've found many people having questions on moving the machine while it's in halt.

One way would be to DIY by disabling the code blocks (if there are separate block) and restarting.
A better way would be to have an advanced pause functionality. 

Issues related to this one where I couldn't find they were solved:
- https://github.com/vlachoudis/bCNC/issues/805
- https://discuss.inventables.com/t/restart-gcode-mid-way/40283/23

How can I check and reinsert the end-mill and perform a z-origin operation in a job that takes +20hours.

Another nice thing with this would be:
1. if you pause you can save the current state, shutdown the machine.
2. you can come back the next day, power up the machine
3. do a homing procedure
4. resume where you left of previously.

Harvie, 2021-09-17:
> How can I check and reinsert the end-mill and perform a z-origin operation in a job that takes +20hours.

Why don't you add manual toolchange in your g-code? that does exactly what you want. pauses operation, waits for endmill to be handled manualy and manualy resumed.

MARIOBASZ, 2021-09-21:
I think what you want to use is Safety Door.
I'm just trying to understand it, in my test I am with a DUE and grblHal, in one pin map it gave me an error, in another it works with a physical button: it stops the axes and the spindle. It seems that he has not gone to Safety Parking, and then the other way around. I will try a grbl.

MARIOBASZ, 2021-09-21:
It seems to work both with physical button and software (tested without physical machine):
1-Raise Z to safe height (default G53 Z-5.00)
2-Stop the spindle
When resuming, turn on the spindle, wait the set time and continue the work
To establish it by soft from bCNC it would be necessary to generate a button that continues operating when the machine is working, and another button that sends ~ to resume.