vlachoudis/bCNC #1624: Tool number and current TLO display on the control page

JipeB, 2021-10-03:
Hello everyone,
I regularly use the manual tool change (TLO) and would like the number of the tool in use and the corresponding TLO displayed on the control page.
On the version I am using (0.9.14.317) only the feedrate is displayed during machining.

I did some research in the code, but I can't seem to locate the routine that allows this display. If anyone could help me.

[image]

JP

rschell, 2022-10-03:
I was noticing this same issue, there are multiple problems with the status frame.  

Most significant one is that the parsing of the TLO response from GRBL (grblHAL in my case) is that some controllers respond with one float for just the z axis and others respond with 3 floats (presumably for XYZ coodinates).  I changed my GRBL1.py to pickoff the last one using the -1 python index convention.  Not sure what happens on an ABC machine, need help there.

Second issue is that the TLO value that a user enters into the input field is not stored in a global place (CNC.vars["TLO"] for example).  Doing so, entries in the Probe frame get reflected in the Status frame.  The Entry field should also be populated with the global value on initialization.  Not sure if the TLO value should be stored in the ini files or not.

On the Tool entry, it is currently ignored and not stored anywhere.  As a starting point, I have placed user entry in CNC.vars["tool"].  I still need work though how and when this value gets used.

JipeB, 2022-10-04:
The notation "TLO" is ambiguous since it denotes both the tool length offset and...the height of the probe.
As far as I'm concerned, only the height of the probe needs to be stored to be available when determining the Program Origin on the stock.
The height of my probe is integrated in a "macro" which shifts the Z0 after initialization; it's very fast and it avoids having to re-enter its value each time.

JipeB, 2022-10-04:
To complete my answer, I'm using FreeCAD to generate the GCode and I modified the Grbl post processor to display the current tool references in the bCNC status banner.
On the other hand I did not find the solution to display the TLO of this tool.
JP

rschell, 2022-10-04:
Even after making my changes to TLO, I suspect the TLO change option does not reflect the new tool length.  It's ineffective at changing the TLO value and results at just resetting the TLO to 0.0.

The WCS tool change option appears to function as I would expect.

JipeB, 2022-10-05:
I don't use the WCS option because it shifts the Z reference at each tool change; This poses a problem if one of the tools breaks. As the reference is no longer the original one, the entire machining must be restarted instead of restarting only the operations corresponding to the replaced tool.
The TLO option, which I use, works correctly provided you add the cancellation of the offset (G49) when calling the tool change procedure.
The TLO option has the advantage of not modifying the Z0 of the program origin and of using the real length of the tool, unlike the WCS option which calculates a difference in length between the old and the new tool.
With this method the length of the tool is calculated automatically; there is no TLO to enter manually.
If you wish, I could tell you the modification to be made to the source file and the progress of the initialization procedure for the change of tool.
JP

rschell, 2022-10-05:
I would like to see your changes, I can seam to get the TLO option perform as one would expect.  The probe of the new tool doesn't change the TLO value correctly for me.

JipeB, 2022-10-06:
Here is the modification I made to the CNC.py file: 2 lines to be inserted at the position indicated by ">>"

lines.append("G49") # reset TLO The current TLO must be canceled before mounting a new tool.
lines.append("%wait") Waiting for cancellation confirmation from GRBL

Function G49 cancels the TLO correction in progress (TLO is set to 0). A new TLO will be created when the next tool is mounted.
#----------------------------------------------------------------------
# code to change manually tool
#----------------------------------------------------------------------
def toolChange(self, tool=None):
	if tool is not None:
		# Force a change
		self.tool = tool
		self._lastTool = None

	# check if it is the same tool
	if self.tool is None or self.tool == self._lastTool: return []

	# create the necessary code
	lines = []
	lines.append("$g")	# remember state and populate variables
	lines.append("m5")	# stop spindle
	lines.append("%wait")
>>	lines.append("G49")	# reset TLO
>>	lines.append("%wait")
	lines.append("%_x,_y,_z = wx,wy,wz")	# remember position
	lines.append("g53 g0 z[toolchangez]")
	lines.append("g53 g0 x[toolchangex] y[toolchangey]")
	lines.append("%wait")

After modifying the CNC.py file:

On Windows:
In command mode, go to the bCNC directory and type pip install bCNC (do not add upgrade so as not to overwrite the modifications)
 
Under Linux: Simply open the files to be modified in “Admin” mode; they will be taken into account the next time bCNC is started.

For the implementation of the procedure: 
[https://www.youtube.com/watch?v=4QMUFPKhjys ](url)
It's in French, but the text can be easily translated

As specified in the video, the machine must be equipped with sensors on the three axes to determine the measurement origin (homing). This is imperative since the tool change procedure is carried out in Machine Coordinates (MCS).

As the measurements taken at the time of calibration are stored in the .bCNC file, they are immediately available at the end of the homing procedure. These measures are operational as long as the table is not resurfaced.

Last remark, the Z0 of the Program Origin (OP) must be determined with the bare spindle, without clamp or tool mounted, under the same conditions as the calibration.

I hope I was clear, if not do not hesitate to contact me if you would like further information.
JP

rschell, 2022-10-06:
Thanks.  I'll give these a try.

rschell, 2022-10-09:
No matter what I try, the purge controller function at the end of the tool change probing macro, resets the TLO to zero.  I have simplified the macro to a file with just four statements in it:

%global TLO; TLO=5.0
G43.1z[TLO]
%update TLO
$#

When it is run, it works at changing the TLO as demonstrated by a parameter list but the purgecontroller function at the end of the run clears the TLO back to zero.  I can enter those same 4 lines individually in the command line successfully.  

I see that the G and TLO restoration codes of purgecontroller are queued but the lines queued after the crtl-x are not sent.  There is no echo of those statements in the terminal log. The problem I believe is that  grblHAL's response to a ctrl-X is:

GrblHAL 1.1f ['$' or '$HELP' for help]

without a subsequent "ok", so the G and TLO restoration lines never get sent.  So I tried using the response with a line containing "grblHAL" to be equivalent to "ok".  That gets the TLO option working.  Will need to review grblHAL code to see if an "ok" response can be added.

JipeB, 2022-10-10:
It's not going to be easy to find help from the developer team that currently maintains the code. It seems that the compatibility between bCNC and grblHal is not total and that it is necessary to rewrite some procedures.
Good luck

rschell, 2022-10-10:
I have replaced my Openbuilds Blackbox (grbl 1.1g) with the grblHAL on a Teensy4.1 board, so I can confirm any fundimental problems and potential solutions.  Right now I can't trust the TLO option because its not persistent.  Too many actions result in the value being lost.  

I just checked the Blackbox and confirm that the TLO is being lost on it too during a ctrl-x.

JipeB, 2022-10-11:
Ctrl-X >> Reset.
The loss of the TLO on a reset is normal since the machine is reset. This will require redoing the Machine Origins and restarting the machining cycle; therefore the assembly of the first tool to which a new TLO will be associated.
A TLO is only active for the duration of a tool action (exactly like tool radius compensation), hence the need to cancel it before mounting a new tool.
grbl, regardless of version, has always handled the TLO this way; nothing really new since the developer of grbl has taken over the basic procedures of industrial computers.
To have persistent TLOs, you need a machine equipped with a tool magazine managed by a database where each tool is archived with its parameters (diameter, length, etc.). Nothing in common with the amateur machines we use.