vlachoudis/bCNC #1647: Probe / Change Tool Auto Zero Help,

technothugjp, 2021-11-21:
Hi Folks 

i make my own cnc and i use GRBL latest version with ARduino and Cnc shield V2 

I make 2 probes for use also, an Leveling probe and a fixed probe for Auto Zero after manual tool change 

https://user-images.githubusercontent.com/94772647/142773180-0e4dd0b9-f130-4ad3-97ac-4567cd3e5148.mp4

https://user-images.githubusercontent.com/94772647/142773181-c671ad39-5322-462b-b6c4-7f5ca1640f0b.mp4

i have some questions about probe, guys, idk how to use some fields of probe

[image]
[image]

i have an issues with my Automatic Zero,

i have added hard limits to my cnc for zero the Mpos and use cordinates to locate my fixed probe for Auto zero, 

idk how to use Calibrate on Tool tabe, 

and now about the Auto zero tool Cycle,

when i start the cycle  its ok,  the tool come to Mposition to manual change
after this i click to resume, and the machine come to probe location and go down to click on probe, full correctly

but when i resume again to tool come back to work the tool back on the same Z move of the last tool, not on the zero of wpos

thx so much for all suport folks,

JipeB, 2021-11-27:
Hello,
Watch this video for manual tool change (TLO). If it matches what you want I could give you some additional information to avoid offset errors.
[(https://www.youtube.com/watch?v=4QMUFPKhjys)]

As far as auto leveling goes, I wouldn't be of any help; I don't use it for my work.

JP

technothugjp, 2021-11-27:
thx for share, i have view on Grbl Gcode comand suport, and Grbl doesnt support gcode M6, that is the issue, but the solution is using macro for do this, thx for share this vid, and try to help me with this,