vlachoudis/bCNC #1648: camera doesn't work in python3

bvarner, 2021-11-22:
I have a small USB camera, works fine with opencv, works fine with V4L2.

When I run bCNC on my machine with a python2 install, the camera works.
When I run bCNC on the _same_ machine with a python3 install, the camera does not work.

Harvie, 2021-11-22:
Do you have opencv module installed for python3 ?
You might only have it installed in python2.

jsiddall, 2022-02-22:
IIRC my camera started working when I installed some PIL OS packages.  Ex:

sudo apt-get install python3-tk python-pip python3-pil python3-pil.imagetk

jensi91, 2024-11-16:
Hello,
I also had the problem that the camera symbol remained grayed out. The following is the problem with the Python installation: nympy version 2.x is installed, but bCNC is designed for 1.x. This can be recognized by a message in the terminal.
I was able to solve the problem by uninstalling nympy and reinstalling it with version 1.19.5:

py -m pip uninstall numpy

py -m pip install numpy==1.19.5