vlachoudis/bCNC #1652: problem display bCNC

pat-autocom, 2021-12-11:
I have a display problem with the gcode (extension ngc) created by
inkscape+gcodetools, the image is not displayed in bCNC although the
code is present in the console.
the inkscape version is 0.91

JipeB, 2021-12-12:
I think some Gcode keywords are not recognized by bCNC. Is the Gcode generated by Inscape compatible with bCNC?

pat-autocom, 2021-12-13:
> Reply-to: vlachoudis/bCNC <reply+AW4GZJQHQTBWEP4DF3KYUUV7YHF3VEVBNHHEAO

Hi, 
Curiously the code is present in the editor, but not the display.
-------- Message initial --------

JipeB, 2021-12-13:
Is this your first time using Inkscape or is it a new version that doesn't work?

Is it machining with complex shapes or basic work?

The editor only displays the contents of the file. It is the interpretation of the code which is contained therein which poses a problem for bCNC to draw the graphic paths. Contrary to what one might think, the standardized Gcodes are not all interpreted by the postprocessors; hence the previous questions.

MARIOBASZ, 2021-12-13:
Maybe very large code, the canvas does not show the entire route, however bCNC can send the g code.
Giving it more time to visualize (button at the top right) might help.

JipeB, 2021-12-14:
The largest code I uploaded to bCNC had over 8000 lines. It didn't pose any problem, but the code had been generated by FreeCAD which has a post processor dedicated to bCNC.
Maybe an Inkscape user could tell us more.

pat-autocom, 2021-12-14:
I changed Inkscape 0.91 to 0.92, now everything works.
Thanks for you help.

-------- Message initial --------

Répondre à: vlachoudis/bCNC <

Date: Tue, 14 Dec 2021 05:06:49 -0800

"target": "
https://github.com/vlachoudis/bCNC/issues/1652#issuecomment-993520420",
"url": "
https://github.com/vlachoudis/bCNC/issues/1652#issuecomment-993520420",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]