vlachoudis/bCNC #1653: bCNC no more working since upgrade to python3.10

fume68, 2021-12-14:
Since the last Update on Archlinux bcnc didnt work anymore. When I try to install I got this Error:

  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python /usr/lib/python3.10/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-g20utw44/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip 'numpy==1.13.3; python_version=='"'"'3.6'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"'' 'numpy==1.19.3; python_version>='"'"'3.9'"'"''
       cwd: None
  Complete output (1045 lines):
  Ignoring numpy: markers 'python_version == "3.6"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8"' don't match your environment
  Collecting setuptools
    Downloading setuptools-59.6.0-py3-none-any.whl (952 kB)
  Collecting wheel
    Using cached wheel-0.37.0-py2.py3-none-any.whl (35 kB)
  Collecting scikit-build
    Downloading scikit_build-0.12.0-py2.py3-none-any.whl (73 kB)
  Collecting cmake
    Downloading cmake-3.22.1-py2.py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (22.7 MB)
  Collecting pip
    Downloading pip-21.3.1-py3-none-any.whl (1.7 MB)
  Collecting numpy==1.19.3
    Downloading numpy-1.19.3.zip (7.3 MB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
      Preparing wheel metadata: started
      Preparing wheel metadata: finished with status 'done'
  Collecting distro
    Downloading distro-1.6.0-py2.py3-none-any.whl (19 kB)
  Collecting packaging
    Downloading packaging-21.3-py3-none-any.whl (40 kB)
  Collecting pyparsing!=3.0.5,>=2.0.2
    Downloading pyparsing-3.0.6-py3-none-any.whl (97 kB)
  Building wheels for collected packages: numpy
    Building wheel for numpy (PEP 517): started
    Building wheel for numpy (PEP 517): still running...
    Building wheel for numpy (PEP 517): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python /usr/lib/python3.10/site-packages/pep517/in_process/_in_process.py build_wheel /tmp/tmph9elqn77
         cwd: /tmp/pip-install-1tcog7ee/numpy_745f98d2654a443a9275aed3d7af4160
    Complete output (1006 lines):
    setup.py:67: RuntimeWarning: NumPy 1.19.3 may not yet support Python 3.10.
      warnings.warn(
    Running from numpy source directory.
    /tmp/pip-install-1tcog7ee/numpy_745f98d2654a443a9275aed3d7af4160/tools/cythonize.py:67: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
      from distutils.version import LooseVersion
    numpy/random/_bounded_integers.pxd.in has not changed
    Processing numpy/random/_bounded_integers.pyx
    numpy/random/mtrand.pyx has not changed
    numpy/random/_pcg64.pyx has not changed
    numpy/random/bit_generator.pyx has not changed
    numpy/random/_sfc64.pyx has not changed
    numpy/random/_common.pyx has not changed
    numpy/random/_bounded_integers.pyx.in has not changed
    numpy/random/_philox.pyx has not changed
    numpy/random/_generator.pyx has not changed
    numpy/random/_mt19937.pyx has not changed
    Cythonizing sources
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE
  
    blis_info:
      libraries blis not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/']
      NOT AVAILABLE
  
    openblas_info:
      libraries openblas not found in ['/usr/local/lib', '/usr/lib64', '/usr/lib', '/usr/lib/']
[... 958 lines cut ...]
    gcc: numpy/core/src/npymath/halffloat.c
    gcc: numpy/core/src/common/array_assign.c
    gcc: numpy/core/src/umath/override.c
    gcc: numpy/core/src/common/mem_overlap.c
    gcc: numpy/core/src/npymath/npy_math.c
    gcc: numpy/core/src/umath/extobj.c
    gcc: build/src.linux-x86_64-3.10/numpy/core/src/umath/scalarmath.c
    gcc: numpy/core/src/common/ucsnarrow.c
    gcc: numpy/core/src/common/npy_longdouble.c
    gcc: numpy/core/src/common/ufunc_override.c
    gcc: build/src.linux-x86_64-3.10/numpy/core/src/common/npy_cpu_features.c
    gcc: numpy/core/src/common/numpyos.c
    gcc: numpy/core/src/common/cblasfuncs.c
    gcc: numpy/core/src/common/python_xerbla.c
    gcc: numpy/core/src/multiarray/mapping.c
    gcc: numpy/core/src/multiarray/methods.c
    gcc: build/src.linux-x86_64-3.10/numpy/core/src/umath/matmul.c
    gcc: build/src.linux-x86_64-3.10/numpy/core/src/umath/clip.c
    error: Command "gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fPIC -DNPY_INTERNAL_BUILD=1 -DHAVE_NPY_CONFIG_H=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -DNO_ATLAS_INFO=1 -DHAVE_CBLAS -I/usr/local/include -I/usr/include -Ibuild/src.linux-x86_64-3.10/numpy/core/src/umath -Ibuild/src.linux-x86_64-3.10/numpy/core/src/npymath -Ibuild/src.linux-x86_64-3.10/numpy/core/src/common -Inumpy/core/include -Ibuild/src.linux-x86_64-3.10/numpy/core/include/numpy -Inumpy/core/src/common -Inumpy/core/src -Inumpy/core -Inumpy/core/src/npymath -Inumpy/core/src/multiarray -Inumpy/core/src/umath -Inumpy/core/src/npysort -I/usr/include/python3.10 -Ibuild/src.linux-x86_64-3.10/numpy/core/src/common -Ibuild/src.linux-x86_64-3.10/numpy/core/src/npymath -c build/src.linux-x86_64-3.10/numpy/core/src/multiarray/scalartypes.c -o build/temp.linux-x86_64-3.10/build/src.linux-x86_64-3.10/numpy/core/src/multiarray/scalartypes.o -MMD -MF build/temp.linux-x86_64-3.10/build/src.linux-x86_64-3.10/numpy/core/src/multiarray/scalartypes.o.d -std=c99" failed with exit status 1
    ----------------------------------------
    ERROR: Failed building wheel for numpy
  Failed to build numpy
  ERROR: Could not build wheels for numpy which use PEP 517 and cannot be installed directly
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/30/46/821920986c7ce5bae5518c1d490e520a9ab4cef51e3e54e35094dadf0d68/opencv-python-4.4.0.46.tar.gz#sha256=d80db278a07f51811dbf0f9c31ff7cd5b2501822fb7a7587e71f9ff27d5c04bd (from https://pypi.org/simple/opencv-python/) (requires-python:>=3.6). Command errored out with exit status 1: /usr/bin/python /usr/lib/python3.10/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-g20utw44/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip 'numpy==1.13.3; python_version=='"'"'3.6'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"'' 'numpy==1.19.3; python_version>='"'"'3.9'"'"'' Check the logs for full command output.

griffintwosix, 2022-01-02:
Well Manjaro just fell too.  So I'm loving bCNC, but now I'm stuck.  Guess I can reload a stale version of linux.  Like Debian or Mint.  Well hope devs get it going, or try make an AppImage of it.  Self contained package would be great.

Crind711, 2022-01-04:
I found a work around until dev get it fixed. Im running manjaro and notice in /usr/lib/python3.10/site-packages that it was missing files needed to run bCNC. I copied and merged all files in /usr/lib/python3.9/site-packages to /usr/lib/python3.10/site-packages. Once its completes- use same command as before to run bCNC. 

I too would like to for developers to make a appimage that is self contained. I would, but Im just a dumb noob! All looks foreign to me!

Harvie, 2022-01-19:
For me it does work on Python 3.10.1

bosd, 2022-06-29:
Is this one resolved?

Crind711, 2022-06-30:
> Is this one resolved?

Not sure it has. I do know what I mentioned above works. I moved back to LMDE 5 and windows 10, Had same issue on windows with the update of python.