vlachoudis/bCNC #1658: Z-probe - G92 setting isn't persistent

mj8w, 2022-01-18:
I have created a new button "Z-Probe". I want it to touch the tool off to a 15mm thick contact.

It has this g-code:

G38.2 Z-25 F10;
G92 Z15;
G90 G0 Z25.0

This appears to work correctly, in that I can run a job with it. 

But if I have a follow-on job with the same tool, **it loses the z-height**. This means that the second job always removes more material than it was supposed to.

I see that when I run the Z-Probe menu item, it sets some settings in the list of workspaces. I'm not sure what it is actually doing in the gcode it produces, or how to duplicate it in my script.

Probing via the menu is harder, I have to change menus and then press the "Probe" button, and then raise the tool afterward. I was really hoping it would be easy to use the G38.2 command from the script.

MARIOBASZ, 2022-01-18:
#239

MARIOBASZ, 2022-01-18:
https://github.com/vlachoudis/bCNC/issues/1658

MARIOBASZ, 2022-01-18:
G92 isnt persistent

jsiddall, 2022-02-23:
bCNC variables are persistent for as long as bCNC is running.  You could store your Z-probe value in a variable and then have a button to re-set G92 manually after each job.

Unfortunately, user buttons are also broken currently (see #1674) but a fix for that is in the works