vlachoudis/bCNC #1661: How to Call any other executable using bCNC custom Buttons

engrjawadulhassan, 2022-01-23:
Hello,

Can i call any other executable from bCNC ? using custom buttons , e.g calculator.exe etc.

I was curious about this , as i read the previous issues but found nothing about this.

Thanks

CmdrCody51, 2022-01-23:
I hope so too. My 3D printer software can (OctoPrint) and I use simple scripts to power lights and printers.
It would be nice to be able in bCNC - turn on lights, mill, vacuum and fans that aren't really specific to GCode.

CmdrCody51, 2022-01-26:
Looks like I can do what I want through the Pendent web ui. At least now I have time to see if I can find it in the program.

Harvie, 2022-01-26:
You should be able to embed python scripts in the buttons g-code. See wiki for details.
https://github.com/vlachoudis/bCNC/wiki/Scripting

CmdrCody51, 2022-01-26:
I sure am glad you verified this before closing it!!!!!
Too bad the example on your wiki doesn't work huh?
I put this in button 9:
%import os; os.system("xterm")
Clicked Ok and it sat there fat dumb and happy.
Guess that makes it more of a BUG.

Harvie, 2022-01-27:
> I sure am glad you verified this before closing it!!!!!

I am sure glad that you've prepared PR with fix instead of pointing fingers huh :-)
It's not like we are paid or otherwise obliged to check your use case. Just trying to provide free software in our personal free time. bCNC has lots of features and when something gets broken it might not be immediately obvious.

Can **please** you take a look why this is not working? I remember it was working ok in some older version.

CmdrCody51, 2022-01-30:
Sorry - ya - I got miffed when every thing says it oughta work and it didn't...
First thing I noticed is my version is 0.9.14-dev on RPi with 3.9.2 python.
It was some fix for running on the RPi. Not sure why but I don't even have some of the tabs.
Looks like I have to drill down into my local copy to find what it ought to do as well as why it won't do it.

CmdrCody51, 2022-01-30:
I see where it oughta run the command.(CNC.py) I guess the editor is croaking on entering the command because it never gets that far.

CmdrCody51, 2022-01-31:
It is choking in configparser.py in the /usr/lib/python3.9/

CmdrCody51, 2022-01-31:
Ah man....................
On the first entry in the command window you have to escape the '%' - so you use '**%%**'
If you edit it later you have to use '**%%**'
It is even stored as '**%%**'
You can close this now.....

Harvie, 2022-01-31:
That's it? Just enter % twice and it works again?
Perhaps we should fix this so it gets escaped automaticaly?

it should be also possible to enter python commands using % in g-code files. i think the syntax should be consistent across bCNC.
Also we should mention this in wiki if we decide to leave it unfixed. But perhaps we should fix it anyway...