vlachoudis/bCNC #1671: Limit switch operation

nealre, 2022-02-02:
I have setup $22=1 and $21=1 $23=0 and $32=1 for my 2.5 watt laser. I am only working with X and Y axes. When I try to go homing cycle the X and Y move toward their respective Homing Switch. When they make contact the alarm goes off but the stepper keeps trying to move. Also, the Homing switch makes the USB loose it's connection? I have to move the stepper away from the sw before I can reset the USB port. I have looked in the config.h file and the homing is setup as well as X,Y,Z steppers. I added a Z stepper and sw just to see it that would correct the problem? but, no, It didn't change anything. The Z stepper does not act right either as it does not turn until it hits the limit sw it just bobbles back and forth and stops. It can use the Origin to set home and it will work as long as the limit switches are not touched. I also know and fixed the D12 limit sw pin.
thanks for your help
Rick

nealre, 2022-02-09:
after wiring the limit switches to normally closed and setting the direction settings $3 to 2 for the Y axes the limit switches work fine but I still can't get the $22=1, $21=1 to work. The soft limits work just fine?
The firmware says they should hit the limit sw and bounce back $27=5.0 but it hits and sets the alarm and keeps trying to move in the home direction? Does anyone have any ideas?

jsiddall, 2022-02-16:
Limit switches work great for me.  My switches are at max X, Y and Z (on my machine the top right rear).  Out of curiosity are you using simple switches or the 4-wire ones with LEDs?