vlachoudis/bCNC #1674: Cannot add code that starts with % to user buttons

jsiddall, 2022-02-16:
I am trying to setup some user buttons on a new bCNC install based on some buttons I had working on another bCNC setup.  The button is simple but the current master (0.9.14.318) won't even let me click "OK" when pasting the code into the command field.  Ex: it will accept this:
```
G53 G0 Z[_newz]
```

But not this:
```
%_newz=-78.2+float(app.entry("Enter stock thickness"))
G53 G0 Z[_newz]
```

Scratching my head about why I can't get something so simple to work given it had been working great before.  Any advice would be appreciated!

jsiddall, 2022-02-16:
Update: I tried running this from CLI and discovered a traceback:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/bCNC/Utils.py", line 526, in __call__
    return self.func(*args)
  File "/usr/local/lib/python3.8/dist-packages/bCNC/Utils.py", line 860, in ok
    config.set("Buttons", "command.%d"%(n), self.command.get("1.0",END).strip())
  File "/usr/lib/python3.8/configparser.py", line 1201, in set
    super().set(section, option, value)
  File "/usr/lib/python3.8/configparser.py", line 894, in set
    value = self._interpolation.before_set(self, section, option,
  File "/usr/lib/python3.8/configparser.py", line 402, in before_set
    raise ValueError("invalid interpolation syntax in %r at "
ValueError: invalid interpolation syntax in '%_newz=-78.2+float(app.entry("Enter stock thickness"))\nG53 G0 Z[_newz]' at position 0
```
So, I had a look at configparser.py and found this note:

```
If a user needs to use a bare % in a configuration file, she can escape it by writing %%. Other % usage is considered a user error and raises `InterpolationSyntaxError'.
```

So I added an extra % in front and it accepted the command.  But then when I ran it I got `Name 'basestring' is not defined` and `Name '_newz' is not defined` so clearly something is broken.

Unfortunately fixing this is above my head.

GitHubCaps, 2022-02-19:
I have a fix for this, but just getting back to the working on this project. @DodoLaSaumure if nothing else, I will attach something here for you! Haha, sorry, just been real busy...

PS: This has nothing to do with #1405, but changes in Python3 configparser, it handles the code differently, so this isn't on our part!

jsiddall, 2022-02-22:
Thanks @GitHubCaps

Sorry to hear you are so busy, but glad to hear you have a fix.  I had looked at #1352 but since it was closed I figured I was having a different issue!

jsiddall, 2022-02-22:
One more bit of info: The system I had this working on before was running Ubuntu 18.04 so probably python2.  Makes sense that this broke with the python3 configparser.

GitHubCaps, 2022-02-22:
> I had looked at https://github.com/vlachoudis/bCNC/issues/1352 but since it was closed I figured I was having a different issue!

@jsiddall The author closed it (because they could not get the results they were after) before I got a chance to implement the actual fix for the initial issue. I meant to do a push (or reopen) at that time, but got sidetracked!

> Makes sense that this broke with the python3 configparser.

Yes, indeed!

I should mention this in the docs for macros/buttons/shortcuts, but for those of us who like to keep our strings short, one thing that no longer works with 3, is omitting spaces around the `modulo`. Now, you have to use spaces, otherwise it is picked up as a command.

use to work:
`"%s"%(some_variable)`

now enforced:
 `"%s" % (some_variable)`

Also, that _thing_ with having to use `%%` for 3 and `%` for 2. Well, at first I was adding and stripping in different places until I found a better solution and in the process I accidentally (lol) made 2 the same as 3. So know the unicode (saving/writing) that didn't work in 2 now works through a monkey patch.

jsiddall, 2023-07-13:
Tested with bCNC 0.9.15 and this seems to be fixed.  Note that every % needs to be replaced with %%.

Guessing this got fixed with #1829 so closing this item.

closed 2023-07-13 by jsiddall