vlachoudis/bCNC #1693: Trace outline of loaded gcode file

Mungbeanz, 2022-03-22:
Is there a way to trace the outline of the loaded gcode file using the "user configurable buttons".
Min X, Min Y
Max X, Min Y
Max X, Max Y
Min X, Max Y
Back to home.

I am basically trying to replicate the custom button in LaserGRBL.  They achieve this by using variables:

left - Left margin of loaded image (X)
right - Right margin of loaded image (X)
top - Top margin of loaded image (Y)
bottom - Bottom margin of loaded image (Y)

https://lasergrbl.com/usage/custom-buttons/

I am guessing a python script is necessary

MARIOBASZ, 2022-03-23:
This?
[image]

MARIOBASZ, 2022-03-23:
[image]

MARIOBASZ, 2022-03-23:
[image]

MARIOBASZ, 2022-03-23:
[image]

Mungbeanz, 2022-03-27:
@MARIOBASZ 

Provided me this solution.  Works perfectly!  Thanks dude!

G90
G0 X[xmin] Y[ymin]
G0 X[xmax]
G0 Y[ymax]
G0 X[xmin]
G0 Y[ymin]

Can alter to include M3 commands and G1 feedrate overrides.