vlachoudis/bCNC #1715: P code of G2/G3 circular interpolation

kimstik, 2022-06-28:
as per https://www.linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3
following code should do 2 turns of helix:
```
G21 G90 G17
G0 X0 Y10 Z0
G2 Z-10 I0 J-10 P2
```

but only one shown
[image]

MARIOBASZ, 2022-06-29:
P is accepted in grbl ?

kimstik, 2022-06-29:
;( you are right. sadly it is ignored and used as P=1

MARIOBASZ, 2022-06-29:
one option is that from bcnc interpret P for g2 g3
another option is to use helical plugin

kimstik, 2022-06-29:
What is helical plugin? Sounds interesting.

MARIOBASZ, 2022-06-29:
[image]

MARIOBASZ, 2022-06-29:
[image]