vlachoudis/bCNC #1719: Break compatibility with python 2

m1ch, 2022-07-10:
Python 2 is EOL for some time now. I propose to remove all code fragments that are necessary for the compatibility with python 2. 

e.g: 
```
try:
    import Tkinter
except ImportError:
    import tkinter
```

Also relates to #1480

bosd, 2022-07-11:
It has been discussed a couple of times before.
Somehwere around last year, the decision has been made to still support python2.

We are a year later now, I would also like to propose to drop the support of python2.
Maybe a release can be made of the current repo, as the last python2 compatible version.
In that way it will always be accessible for those who need it.

Then we can continue development and improve the code for python3.

tatarize, 2023-01-02:
https://github.com/vlachoudis/bCNC/commit/6905d8ea64c5f9e6fe78a587e18558db98506bcf

After this PR support for py2 is should probably be officially dead.

I noticed that my addition to svg_elements.py now included fstrings. This means functionally they are Python 3.6+ only. I generally had a couple cases of 2/3 compat code.