vlachoudis/bCNC #1726: Wrong G code parsing

muhammadelmogy, 2022-07-28:
I have generated some G code with [Flatcam](http://flatcam.org/) and as usual opened it with bCNC and it did not show the complete design.
At first I did not really notice it(even it was very obvious that there was something missing in the design) but I went through the process of engraving the design and in the middle of the engraving I noticed that the design was not complete 🤣 (was not good moment at all)

I have tried to open the design files from other programs for example online G-code viewers, ioSender and others and they all opened the complete design without any missing features.
I have even tried to open the G-code file from 2 different PCs (raspberry pi4 & windows) using bCNC and I got the same issue
after that I tried other design files and I faced the same issue

Files used:
[copper_top.gbr_0.2214_iso_cnc.zip](https://github.com/vlachoudis/bCNC/files/9211462/copper_top.gbr_0.2214_iso_cnc.zip)
[isolate.zip](https://github.com/vlachoudis/bCNC/files/9210994/isolate.zip)

rschell, 2022-07-29:
Try changing the timeout setting in the upper right corner of the editor frame.  It appears its taking too long to render the drawing using 5 seconds.  On a Pi 4b machine 30 seconds might be enough.

muhammadelmogy, 2022-07-29:
It did work, but this is really very bad design choose to do it like this!!!!
Why put time out like that while the parser still working, why just not tell the user to wait until the parser finish or at least till the user a timeout has occurred and the parser did not parse all the opened file!!!!!
How to allow the user to start the job with such a huge error in the design G-code load!!

And why it is taking too much time in the first place while other software open it in a fraction of a second!

rschell, 2022-07-29:
Sorry, I didn't write the code, just observed what was triggering your issue.  I would agree that having a timeout on such a critical function is bad, my thought would be to eliminate it.

bCNC loads the file correctly, but it does not finish pre-processing the gcode for rendering and milling purposes. 

@Harvie or @vlachoudis  What are the issues that draw_time was trying to solve?  I can understand the drawing the path can take time for large files but a run should not be allowed to start until its completed.  Should the block.resetPath() be done regardless of whether draw_paths is true or not in drawPaths?

muhammadelmogy, 2022-07-29:
> @Harvie or @vlachoudis What are the issues that draw_time was trying to solve? I can understand the drawing the path can take time for large files but a run should not be allowed to start until its [...]

try upload the attachached files here
https://ncviewer.com/
it will load it in the 3D view in no time, also I tried [ioSender ](https://github.com/terjeio/ioSender) and also it load the 3D view in no time, I think bCNC is trying to do something else rather than just loading the gcode in to the 3D view !

rschell, 2022-09-09:
@Harvie or @vlachoudis What are the issues that draw_time was trying to solve? I can understand the drawing the path can take time for large files but a run should not be allowed to start until its completed. Should the block.resetPath() be done regardless of whether draw_paths is true or not in drawPaths?

Is there interest in PR to remove this code?