vlachoudis/bCNC #1728: Black screen on startup

wbendick, 2022-07-29:
### Discussed in https://github.com/vlachoudis/bCNC/discussions/1724

<div type='discussions-op-text'>

<sup>Originally posted by **wbendick** July 27, 2022</sup>
I just installed bCNC on a Mac. When I start it I get a black screen. Any ideas on how to fix this?

[image]
</div>

rschell, 2022-07-30:
don't have bCNC installed on a Mac but this looks like an issue with tkinter code,  Might google "tkinter mac black screen" for solutions.  Might try python 3.10.x

bosd, 2022-09-05:
Have you tried the recent code base? Downloaded from GH?
Does this problem still exsist?