vlachoudis/bCNC #1738: garbage reciev

sugigpi, 2022-08-16:
I have a problem with bcnc, when bcnc is working it suddenly stops in the middle of the process, when I check it turns out that there is a message in the terminal "garbage received" asking how this happened, please ask for a solution, friend, thank you.

rschell, 2022-09-14:
> File "C:\Python27\Lib\site-packages\bCNC\controllers\GRBL1.py", line 206, in parseBracketSquare
> CNC.vars["prby"] = float(word[2])
> ValueError: invalid literal for float(): 0.0000.000

Traceback suggests a formatting issue in the gcode file.

sugigpi, 2022-09-29:
After I installed a new version of python, the problem is still the same, sometimes the error still comes out, but I switch to another usb port sometimes the problem can be solved, but it's really annoying when I'm working. thanks for the advice bro.

sugigpi, 2022-09-29:
> > File "C:\Python27\Lib\site-packages\bCNC\controllers\GRBL1.py", line 206, in parseBracketSquare
> > CNC.vars["prby"] = float(word[2])
> > ValueError: invalid literal for float(): 0.0000.000
> 
> Traceback suggests a formatting issue in the gcode file.

thanks bro i will try

sugigpi, 2022-09-29:
> > File "C:\Python27\Lib\site-packages\bCNC\controllers\GRBL1.py", line 206, in parseBracketSquare
> > CNC.vars["prby"] = float(word[2])
> > ValueError: invalid literal for float(): 0.0000.000
> 
> Traceback suggests a formatting issue in the gcode file.

After I installed a new version of python, the problem is still the same, sometimes the error still comes out, but I switch to another usb port sometimes the problem can be solved, but it's really annoying when I'm working. thanks for the advice bro.

sugigpi, 2022-09-29:
> > File "C:\Python27\Lib\site-packages\bCNC\controllers\GRBL1.py", line 206, in parseBracketSquare
> > CNC.vars["prby"] = float(word[2])
> > ValueError: invalid literal for float(): 0.0000.000
> 
> Traceback suggests a formatting issue in the gcode file.