vlachoudis/bCNC #1742: Main window not opening

bosd, 2022-09-02:
Just downloaded latest code on this repo.
Try to start it on windows 10 python 3.10 environment.

Main window is not opening.
Traceback:

```
================================================================================
WARNING: bCNC was recently ported to only support
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x000001752D772C20>
Traceback (most recent call last):
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\site-packages\bCNC\__main__.py", line 226, in <module>
    main()
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\site-packages\bCNC\__main__.py", line 153, in main
    application = bmain.Application(className=f"  {Utils.__prg__}  ")
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\site-packages\bCNC\bmain.py", line 159, in __init__
    self.tools = Tools(self.gcode)
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\site-packages\bCNC\ToolsPage.py", line 1285, in __init__
    exec(f"import {name}")
  File "<string>", line 1, in <module>
  File "D:\Users\BOSD\AppData\Local\Programs\Python\Python310\lib\site-packages\bCNC\plugins\trochoidPath.py", line 30
    def __init__(self, master):
                               ^
IndentationError: unindent does not match any outer indentation level
```

@m1ch Can you take care of this one? 🙏

rschell, 2022-09-02:
preceding doc record in trochoidPath.py has a tab at the beginning.

bosd, 2022-09-02:
Thanks! Was a quick fix. Created a PR for this.

PR #1743 (2022-09-02): [FIX] Indentation Error
fixes #1742 

@vlachoudis @Harvie 
Please prioratize this one.
closed 2022-09-02 by Harvie