vlachoudis/bCNC #1745: wiki installation section is missing

swarfer, 2022-09-09:
Hiya
I am trying to point someone at the intallation instructions, but they are missing.
when i am not logged into github, the screen flickers and goes back to the index.
When I am logged in, clicking 'installation' takes me to the 'create new page' editor.

Harvie, 2022-09-10:
Anyone with github account can edit the wiki. But i am not sure where do i find the history, so i can roll it back...

rschell, 2022-09-10:
Appears that section was deleted back in July, I just reverted that deletion.

swarfer, 2022-09-10:
Thankyou

Harvie, 2022-09-10:
@rschell thank you. where did you found that? i was not able to find it...

rschell, 2022-09-10:
I went to the Wiki page and clicked edit, then reviewed the history of the page

Harvie, 2022-09-10:
But there was no edit button when the page was not existing :-D

rschell, 2022-09-10:
Think I when to the top level history and looked at the top level history edits (clicked on the History select-able label), it showed up as "Destroyed Installation (markdown)" on Jul 19th.

Looks like this (#1745) and #1725 can be closed now.

closed 2022-09-14 by Harvie