vlachoudis/bCNC #1750: Canvas timeout selection converted to integer

rschell, 2022-09-23:
If the user selects the "inf" entry in the timeout selection box, that entry gets immediately converted to an integer with a value of 300.  

Suggest changing DRAW_TIME global parameter to accept either the string "inf" as entered in the combox box list or an integer.  By delaying its conversion to an integer and adding a trap to make sure DRAW_TIME is not the "inf" string in the timeout test allows an infinite timeout to occur:

                        if DRAW_TIME != 'inf' and time.time() - startTime > int(DRAW_TIME):

rschell, 2022-09-29:
I have a Commit if there is interest in improving this.

Harvie, 2022-09-29:
well.. do you really want to wait for infinite amount of time everytime you click on something in canvas...

Harvie, 2022-09-29:
we need to integrate some accelerated 3D visualisation canvas... #591 ?

rschell, 2022-09-29:
My experience is the timeout can be changed from inf to any other setting during a canvas draw, if the user does not want to wait.

rschell, 2022-10-03:
I guess the other choice would be to remove 'inf' from the list box selection to avoid a mixed mode variable.