vlachoudis/bCNC #1759: Muiltiple issues with new installation of 0.9.14-dev

imechura, 2022-11-10:
Hello,

I installed bCnc via the windows exe installer and it appears to work for the most part but I have run into some issues that are preventing me from moving forward with it.

1) I cannot find a "tools" menu anywhere but it is referenced multiple times in the documentation.
2) I cannot see the controls on the bottom of the screen. If I got to the probe screen there are some number of controls that are cut off at the bottom of the window. It seems like there should have been a scroll bar or something. 
3) There is no bCNC configuration file in my home directory but the documentation refers to one. 
4) Whenever I have tried to use the autolevel function, after clicking the scan button bCNC issues a G0Z3 command which sends my Z Axis to its limit. Changing the Z Min ZMax in the autolevel panel appears to have no effect on this issue. I'm assuming that this might be in the configuration file but see #3 above.

Could these all be related somehow? I am running windows 11 home edition on a brand new laptop. The Cnc is GRBL 1.1 on an Arduino Nano clone. The cnc functioned correctly when using a different sender application.,

imechura, 2022-11-10:
I do not know why the issue is displayed with "import queue error". I did not add that so I guess that github did.

Harvie, 2022-11-10:
I think that only instalation via pip is currently supported:
https://github.com/vlachoudis/bCNC/wiki/Installation

imechura, 2022-11-10:
I uninstalled the .exe version and reinstalled it with PIP and the top level menu "tools" is still not shown in the application. I have not gotten a chance to test the other issues yet. 

Why would there is be a missing top level menu item?

imechura, 2022-11-10:
Also whenever the application is started it prints the following:

Traceback (most recent call last):
  File "C:\Users\imech\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bCNC\ToolsPage.py", line 1099, in __init__
    exec("import %s"%(name))
  File "<string>", line 1, in <module>
  File "C:\Users\imech\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bCNC\plugins\slicemesh.py", line 38, in <module>
    import stl #FIXME: write smaller STL parser
  File "C:\Users\imech\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bCNC\lib\stl\__init__.py", line 1, in <module>
    from .stl import BUFFER_SIZE
  File "C:\Users\imech\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bCNC\lib\stl\stl.py", line 12, in <module>
    from . import base
  File "C:\Users\imech\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bCNC\lib\stl\base.py", line 77, in <module>
    class BaseMesh(logger.Logged, collections.Mapping):
AttributeError: module 'collections' has no attribute 'Mapping'

Harvie, 2022-11-11:
> I uninstalled the .exe version and reinstalled it with PIP and the top level menu "tools" is still not shown in the application. I have not gotten a chance to test the other issues yet.
> 
> Why would there is be a missing top level menu item?

I think it might have been renamed from Tools to CAM

bgbsww, 2022-12-14:
In python 3.10 ( and I think all 3.x )  collections.Mapping doesn't exist.  As far as I can tell, simply removing it as a base class for BaseMesh and getting rid of the vestigial imports solves the problem.