vlachoudis/bCNC #1762: unexpected machine movement and crash

imechura, 2022-11-17:
Hello,

on 3 occasions since I started using bcnc I have had an unexplained sudden movement of the machine which ended in a crash. The first time I suspected it was something I did. The second time I was less convinced that it was me. The third time now I cannot come up with any reason that I could have caused this.

I had just completed a PCB isoaltion routing job using the autolevelling feature. I clicked on file->open to load my XLN file for the drilling operation. bcnc promted me for something, then it said sopmethng in another prompt about autolevling data. 

After I clicked ok, the machine took off on all axis and crashed into my table clamp and broke the end mill. I looked into the terminal and was able to see where my previous job had completed, then I saw that GRBL was reset. Then it showed that for some reason the following command was sent to GRBL G90G0X0.603244Y9.0091. I have no idea where that command came from. My machine only has 4 inches of travel on the Y axis and both the jobs (the one I completed and the one I was opening) are less than 1.5" of travel on the Y axis.

Even if the file did contain that command (which is does not), why would the system just execute it without being instructed to do so?

let me know what information I need to provide if any.

I am running bCNC 0.9.14-dev on windows, installed from PIP. The CNC controller is a GRBL 1.1 on an Arduino Nano.

Thanks for your help.

MARIOBASZ, 2022-11-17:
Do you have any command that indicates G53 and the height in Z?
Autoleveling has that serious problem.
I don't remember if the problem is also generated when the position is outside the polling grid.

MARIOBASZ, 2022-11-17:
For example: Instead of going to G53 Z-3 , the tool goes to Z-3 (compensated by aurolevel) ... but from the current work area!!! (probably G54)
Houston, we're in trouble.

imechura, 2022-11-17:
I had read about an issue where bcnc will rapid to the machine coordinates 0 position if you are running a probe and stop before the probe activates the switch. That is similar to what I am experiencing but I was not probing. I was in the middle of opening a file. Then there was a dialog that said that probe data exists and asked me if I wanted to save it. I answered no then bcnc commanded grbl to rapid the machine until my spindle crashed into something by sending a G0 command.