vlachoudis/bCNC #1766: bCNC wont open

GabriellJohnny, 2022-11-30:
when i try to start the program i get the following message in the prompt and then it crashes

[image]

dont know how to fix it, all the modules are installed correctly , this behavior started all the sudden. I am running the program on the Win10 .

bgbsww, 2022-12-14:
That's reading the configuration file.  It would be ~/.bCNC and <program home>/bCNC.ini on a unix system.

On a windows system, it should be .bCNC in your user's home directory, and bCNC.ini wherever the application is installed.  One or the other of those files is corrupt.

Looking at the stack tract, it's c:\\Users\Volts/.bCNC which is your user config file.  Delete that file, and things should work again.

GabriellJohnny, 2022-12-25:
it worked, tahnk you!

closed 2022-12-25 by GabriellJohnny