vlachoudis/bCNC #1793: Send Gcode one by one

zerox100, 2023-01-03:
Dear,

Thank you for your nice software. I want to send GCodes one by one to the board. Is there any parameters to configure?

Best regards

Harvie, 2023-01-03:
You mean line by line manualy? There is commandline in bCNC, so you can just enter your g-code commands there.

zerox100, 2023-01-03:
I mean that the software send one gcode to the board. After completion of one command the software will send next gcode immediately. 

In other word there is no buffer in the board.

andrewmarles, 2023-01-03:
This is done in the controller:

https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md#c---check-gcode-mode

zerox100, 2023-01-03:
I've checked it. I mean the computer software send the first gcode and wait for controller board to complete the first gcode and then send second gcode just after finishing the previous gcode

Harvie, 2023-01-04:
But why would you do this? It would mess up acceleration/deceleration, since grbl will need to accelerate and fully decelerate during each g-code line which will cause many issues like being slow and having bad surface finish.

zerox100, 2023-01-04:
I see and understand....
I will add the queue and acceleration option. But now i need to simplify the process for debugging purposes.

MARIOBASZ, 2023-01-04:
send "%wait" can help?

zerox100, 2023-01-08:
Nop, %wait could send from bCNC!
not from Controller board

terjeio, 2023-04-08:
[grblHAL](https://github.com/grblHAL) supports single stepping gcode, `$S` toggles single stepping on/off. Send the cycle start real time command (or trigger the cycle start input) to execute the next.