vlachoudis/bCNC #1801: bCNC with fluidNC 3.6.6 - Don't work more.

RDS356, 2023-01-17:
The latest version of FluidNC don't work more with Bcnc.
The FluidNC 3.6.5 and early works very well.
Any idea ?

[image]

Harvie, 2023-01-17:
You have bCNC from git?

RDS356, 2023-01-17:
> You have bCNC from git?

Yes, version 0.9.15 phyton 3.8.7.

rschell, 2023-01-17:
Traceback suggests that the tool field is blank from the G string.  

Could change "int(g(1:)" to "int(g(1:) or 0)"

rschell, 2023-01-17:
I had gone a step further and only process the T code when the tool is managed by controller:

                # accept value only if tool change managed by grbl
                if CNC.toolPolicy <= 1: CNC.vars["tool"] = int(g[1:] or 0)

closed 2023-01-30 by RDS356