vlachoudis/bCNC #1806: Error calculating the minimum distance from arc to point

MARIOBASZ, 2023-01-28:
Will I be wrong?
distance (self,P) for CW and CCW generates the wrong distance to a point: it takes as reference for the calculation A or B of the vector.
Fortunately, discarding options for Line, CW or CCW, seems to solve the measurement error!

LittlePierre, 2023-01-29:
in bpath.py line 644, seems ok
```python
class Segment:
...
...
    # ----------------------------------------------------------------------
    # Return minimum distance of P from segment
    # ----------------------------------------------------------------------
    def distance(self, P):
        if self.type == Segment.LINE:
            AB2 = self.AB[0] ** 2 + self.AB[1] ** 2
            APx = P[0] - self.A[0]
            APy = P[1] - self.A[1]
            if abs(AB2) < EPS:
                return sqrt(APx**2 + APy**2)
            dot = APx * self.AB[0] + APy * self.AB[1]
            proj = dot / AB2
            if proj < 0.0:
                return sqrt(APx**2 + APy**2)
            elif proj > 1.0:
                return sqrt((P[0] - self.B[0]) ** 2 + (P[1] - self.B[1]) ** 2)
            else:
                d = (APx**2 + APy**2) - dot * proj
                if abs(d) < EPS:
                    return 0.0
                return sqrt(d)

        elif self.type == Segment.CW:
            PCx = P[0] - self.C[0]
            PCy = P[1] - self.C[1]
            phi = atan2(PCy, PCx)
            if phi < self.endPhi - EPS / self.radius:
                phi += PI2
            if phi > self.startPhi + EPS / self.radius:
                return sqrt((P[0] - self.A[0]) ** 2 + (P[1] - self.A[1]) ** 2)
            else:
                return abs(sqrt(PCx**2 + PCy**2) - self.radius)

        elif self.type == Segment.CCW:
            PCx = P[0] - self.C[0]
            PCy = P[1] - self.C[1]
            phi = atan2(PCy, PCx)
            if phi < self.startPhi - EPS / self.radius:
                phi += PI2
            if phi > self.endPhi + EPS / self.radius:
                return sqrt((P[0] - self.B[0]) ** 2 + (P[1] - self.B[1]) ** 2)
            else:
                return abs(sqrt(PCx**2 + PCy**2) - self.radius)
```

MARIOBASZ, 2023-01-29:
Hello Pierre.
She had drawn an arc. Measuring distance to the point (0.0, 0.0) gave me the wrong distance: start point or end point, depending on whether it was CW or CCW. If it linearizes, it generates the correct result.
I am making a schematic plugin, my intuition is that from there I can generate an adaptive pocket (each point obtained must have the minimum free space information)