vlachoudis/bCNC #1808: Connxion lost during job

HoRaMusic, 2023-01-30:
Hello, 

I'm trying to use bCNC to make PCBs since many days.
The problem is that during the milling from my gcode the connxion is lost after a few minutes.
I'm using a 3018 PROVer. Since I have read that the CH340 serial converter can create this problem 
I have "hacked" the 3018PROVer to work with an external usb to serial converter based on a FT232RL 
but it didn't resolved the problem.

Is there a way to get it properly working? or a workaround to restart the milling where it was before the connexion problem happened?

rschell, 2023-02-01:
Had a similar issue recently with a new controller configuration.  I added noise suppression ferrites to each end of the USB cable.  My try a pair of those.

vadimyar, 2023-09-08:
I encountered a similar problem for more than a year (3018 Grbl 1.1), I tried everything, including ferrites, but I noticed that if the spindle is not connected, the code runs without communication problems and there has never been a loss of connection between the bCNC and Grbl, not once!, apparently a collector motor and PWM, connected the spindle from a separate power supply, decoupled the PWM and level probe through optocouplers, ordered galvanic isolation of the USB port for Ali, “looked” at the main power bus during the operation of the router for strong impulse noise, pressed with everything possible with condensers, chokes... Last hope for USB ADUM4160/ADUM3160 :)))

Harvie, 2023-09-08:
what does "lost connection" mean? is actual USB disconnect happening (eg. similar report by OS as when you unplug and plug again the usb cable)?

Have you tried shorter usb cable? Also proper shielding of stepper and spindle cabling is important.

vadimyar, 2023-09-11:
> что означает "потерянное соединение"? происходит ли фактическое отключение USB (например, аналогичное сообщение операционной системы при отключении и повторном подключении USB-кабеля)?
> 
> Пробовали ли вы использовать более короткий USB-кабель? Также важно надлежащее экранирование кабелей шагового двигателя и шпинделя.

First of all, many thanks to the developer and support of the project!
I tried everything, including different lengths of the USB cable, I think that the problem is in the hardware (a Chinese clone of the machine), perhaps poor-quality soldering (even a slight vibration affects it, an absolutely random loss of connection) I think somewhere in the area of the USB controller chip CH340.
I don’t give up trying to figure out this unknown crap :)))

vadimyar, 2024-05-08:
SOLUTION. The problem was resolved, after installing galvanic isolation along the USB line, the loss of communication between the computer and the CNC stopped. USB to USB ADUM4160 ADUM3160 
[image]