vlachoudis/bCNC #1811: G-Code path deformed after applying "CUT"

Harold-Fox, 2023-02-13:
I use shx-fonts in Lightburn to create the g-code for engraving labels. Everything worked fine a few weeks ago, but now that I need a few more labels, I follow the same steps whith an unexpected result:
The g-code opens fine in bCNC as usual,  but once I select it and apply CUT, its shape completely changes into a long winding path with curves and slings.
I have tried several times, I always end up the same way now. (I know I have to change the g-code header to make it work for a router)
I can create text in bCNC as well, but it only opens ttf-fons, which is not so handy for engraving.
So does anybody know what went wrong and how to fix it?
Or: Will shx-fonts be allowed to open some time in the future?

Harvie, 2023-02-14:
Can you please provide some screenshots and affected g-code?

Harold-Fox, 2023-02-14:
/media/neuland/USTHREE PRO/Dokumente/Linux CNC/2023-02-14 20-25-01-unselected.png
/media/neuland/USTHREE PRO/Dokumente/Linux CNC/2023-02-14 20-26-59-selected and cut.png
Before and after,
nice piece of artwork, undesired though. I had to scale the result down, it inreases in size after applying CUT

Harold-Fox, 2023-02-14:
[image]
[image]

LittlePierre, 2023-02-16:
Could you please attach the gcode before the cut in order to reproduce ?

Harold-Fox, 2023-02-17:
[Paprika-ger.zip](https://github.com/vlachoudis/bCNC/files/10772199/Paprika-ger.zip)
Here it is, I hope you are successful

LittlePierre, 2023-02-18:
Your gcode is relative. ( G91)
I do not have the fix for the bug right now

However, a solution to your problem is
a) save your gcode to dxf
b) reload the dxf  (It is in G90 now)
c) Order path with 'editor/optimize'
d) apply the cut

Hope this helps
All the best
Pierre

[image]

[Paprika-ger-dxf.zip](https://github.com/vlachoudis/bCNC/files/10773500/Paprika-ger-dxf.zip)

[Paprika-gerg90.zip](https://github.com/vlachoudis/bCNC/files/10773501/Paprika-gerg90.zip)

Harold-Fox, 2023-02-19:
Thank you very much! 
So I will have a look that Lightburn will encode it in G90 right away. Otherwise I am glad you showed me another way.
I will leave a note.....

Harold-Fox, 2023-02-19:
Thank you, that was it. Problem solved.

Harvie, 2023-02-19:
I guess there is still bug in bCNC then right?

Harold-Fox, 2023-02-19:
Not that I know. I changed the setting to "absolute coordinates" in LighBurn as suggested above and created a g-code that worked fine in bCNC. I guess some moment in the past months I changed the settings and forgot about it, not knowing its impact. 
But it would be nice if bCNC could handle SHX files right away. It would save me - and maybe others-  from the detour via LightBurn concernig work with single path fonts.

LittlePierre, 2023-02-20:
> I guess there is still bug in bCNC then right?
@Harvie 
I have analyzed the code quickly

I would talk of missing functionality rather than a bug.
 I have seen plenty of major issues with G91  in the CAM module.
We should close this one that is not a bug (shx fonts), and reopen a new issue "G91 in CAM module" => I think it needs major code changes to be solved....
all the best
Pierre

LittlePierre, 2023-02-20:
From what I see, in CAM module both gCode and paths are intricate since gcode is generated dynamically.

Maybe a clean way would be to manage paths only in CAM Module, then post generate gCode, but it needs a full rewrite of all CAM modules. This would also avoid approximations in paths ( tiny segments are sometimes generated)
But that is not the way bCNC works.

This also applies to pockets, offsets, profiles,  etc...

LittlePierre, 2023-02-20:
> Not that I know. I changed the setting to "absolute coordinates" in LighBurn as suggested above and created a g-code that worked fine in bCNC. I guess some moment in the past months I changed the [...]

@Harold-Fox 
Have you tried to draw text from CAM/generator/text plugin?
I think that shx fonts should be imported there. I guess that this works.( not tried)
Pierre

Harold-Fox, 2023-02-20:
Thank you for your concern. I tried to open shx fonts in bCNC before I opened this thread and it does not work. The original problem would otherwise not have occurred. At least I know the setting for LighBurn to export a working g-code.
I am not a programmer and for me the goal has been achieved. 
Maybe there is someone who writes a plugin for shx fonts for bCNC. Unfortunately I cannot contribute to that.

LittlePierre, 2023-03-03:
@Harold-Fox could you please attach the font (zip file) I would like to understand why it cannot be imported into bCNC. Thanks. Pierre

Harold-Fox, 2023-03-03:
[shx-fonts.zip](https://github.com/vlachoudis/bCNC/files/10884273/shx-fonts.zip)
Of course, here it is. I used the hand.shx font from this collection
Thank you for looking into this

LittlePierre, 2023-03-04:
OK I understand. These fonts are Autocad format fonts. 
They are not supported by bCNC : bCNC supports ttf fonts.
But maybe we could reuse the lib shx parser https://github.com/tatarize/shxparser from @tatarize 
Pierre

tatarize, 2023-03-04:
That parser should be quite solid. I went through a lot of the test cases and different formats. Including the specifically cited files here. And looked through the various tools used by the shx format. If it's not valid it will throw a ShxFontParseError and if it does parse, you can write your own little path code by feeding it a series of characters. As far as formats go it's a bit complex but fairly straight forward to parse. I've also written a pure TTF parser for python and that's a couple orders of magnitude harder.

You need to be able to process straight-line segments and arcs (or just segmentize the arcs) to parse SHX and you need to send in a series of ascii characters to get out your result. It's a proper-ish font format. The files themselves just contain a series of 16 different types of movement commands, and those programs are appended together. So the font will do the typesetting of the font by moving over accordingly.

Oh, and there's a flag that is sent in for `horizontal` vs. vertical fonts. This basically controls a `COND_MODE_2` command that can trigger will often trigger a XY_DISPLACEMENT controlling where the next glyph gets written but could control everything since the commands are basically building a drawing computer program out of sections of code that are evoked by glyphs.

A font would be really hard to write yourself, but processing them is fairly straight-forward. Though it needs all the functions to work correctly, but it's just fetch these commands, then these commands and execute these commands on writing a path.

-- Input, `shx-file`, `text-string`, `horizontal-flag`.
-- Output, path commands of lines and circular arcs.

The test direction in the shxparser-project can give you a solid examples of how effectively the parsing works. But, it is mature enough to be included places and if anything wrong with it, I'd be happy to correct any bugs.

PS. I wrote the svg parser you guys use, and my meerk40t project if pretty mature and has a number of overlaps (though strictly a laser-control project).

LittlePierre, 2023-03-05:
Thanks @tatarize 
With #1814 we should be able to import shx fonts from text plugin.
I have been playing with the horizontal flag, but I could not render a vertical font, maybe I have missed something.

All the best 
Pierre

LittlePierre, 2023-03-05:
Close this one and open #1816

LittlePierre, 2023-03-05:
@Harold-Fox can you please update to the last version and check if you can import shx fonts. Thanks. Pierre

Harold-Fox, 2023-03-05:
I just updated to the new version, but unfortunately I get the same embarassed message that it cannot open this font file.

LittlePierre, 2023-03-06:
This must be because the version on pypi depository dates from 2022 01 24 and has not be refreshed since. 
So when you upgrade you get the old version.

I have opened #1818 to upgrade version to 0.9.14.319

LittlePierre, 2023-03-06:
@Harvie I am sorry, I used a bad import for using shxparser vendor's lib. the code
`from shxparser import ShxFont,ShxPath`
is wrong
The tests went ok because I had a shxparser.py in my lib directory.
But after further tests we need to import
`from shxparser.shxparser import ShxFont,ShxPath`
I have created #1819 and #1818 
Sorry for the confusion
All the best
PIerre

LittlePierre, 2023-03-06:
@Harold-Fox 
this should fix  your issue now
`pip install --upgrade git+https://github.com/vlachoudis/bCNC`
Can you please try it and report the result
Thanks
Pierre

Harold-Fox, 2023-03-06:
I just tried: Message in terminal: successfully installed 0.9-14-319
Still the same message: I cannot open this file.

LittlePierre, 2023-03-06:
Here's what I get
```
~$ pip install --upgrade git+https://github.com/vlachoudis/bCNC
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/vlachoudis/bCNC
  Cloning https://github.com/vlachoudis/bCNC to /tmp/pip-req-build-o4wv4p0m
  Running command git clone --filter=blob:none --quiet https://github.com/vlachoudis/bCNC /tmp/pip-req-build-o4wv4p0m
  Resolved https://github.com/vlachoudis/bCNC to commit 11472b7463a7e84a5396c1c54ccd1e818ecdba35
  Preparing metadata (setup.py) ... done
Requirement already satisfied: Pillow>=4.0 in /usr/lib/python3/dist-packages (from bCNC==0.9.14.319) (9.0.1)
Requirement already satisfied: numpy>=1.12 in /usr/lib/python3/dist-packages (from bCNC==0.9.14.319) (1.21.5)
Collecting shxparser>=0.0.2
  Using cached shxparser-0.0.2-py2.py3-none-any.whl (11 kB)
Collecting svgelements<2,>=1
  Using cached svgelements-1.9.0-py2.py3-none-any.whl (132 kB)
Requirement already satisfied: opencv-python==4.5.5.62 in ./.local/lib/python3.10/site-packages (from bCNC==0.9.14.319) (4.5.5.62)
Requirement already satisfied: pyserial in ./.local/lib/python3.10/site-packages (from bCNC==0.9.14.319) (3.5)
Building wheels for collected packages: bCNC
  Building wheel for bCNC (setup.py) ... done
  Created wheel for bCNC: filename=bCNC-0.9.14.319-py3-none-any.whl size=1404672 sha256=bb59015744b9df235467f6ce25f71071fd2ad18807a121191092ddaefd89cb89
  Stored in directory: /tmp/pip-ephem-wheel-cache-3_20q1pu/wheels/51/9a/db/4b644415538a567e3c3b1fd4f822a907d53120125fbcf78fce
Successfully built bCNC
Installing collected packages: svgelements, shxparser, bCNC
  Attempting uninstall: bCNC
    Found existing installation: bCNC 0.9.14.318
    Uninstalling bCNC-0.9.14.318:
      Successfully uninstalled bCNC-0.9.14.318
Successfully installed bCNC-0.9.14.319 shxparser-0.0.2 svgelements-1.9.0

```

Then running bCNC gives the following

```
~$ python -m bCNC
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x7f31a68be3b0>
Unable to import Image, ImageTk from Pillow

G2Core loaded
self.canvas.winfo_id(): 14680158
Not Loading 6 axis displays
Not Loading 6 axis displays
~$ 
```
I am working with ubuntu 22.04, from ~/ directory it is working fine for me.

Are you running bCNC from a location where there would be  an older copy of bCNC remaining ?

Harold-Fox, 2023-03-06:
The git module was missing. After installation from synaptic it worked.
Goals achieved: Path deformation due to relative settings, now set to absolute, works fine
SHX fonts now available in bCNC, no more need to detour to LightBurn
Thank you all who contributed.

LittlePierre, 2023-03-06:
Great !!
So I think you can now close this issue.
I have opened another one for G91 that does not work in CAM module.
Pierre

closed 2023-03-07 by Harold-Fox